plerup/makeEspArduino #137: show warning

fabiuz7, 2020-05-18:
Hi,
is it possible to show the compiler warnings? In Arduino IDE I have set the flag "compiler warning: all" from the preference panel, so I can see messages such as "warning: unused variables".
I have tried with:

     BUILD_EXTRA_FLAGS:=-Wextra -Wall 

but it doesn't change anything, I cannot see any warning...

plerup, 2020-05-19:
Hrm, just realized that the help text is wrong, the variable name is CPP_EXTRA

Please try that instead

fabiuz7, 2020-05-19:
I would precise that I'm using the version 5.0.2. 
CPP_EXTRA seems doesn't have any effect. Hence it tried to enable VERBOSE flag.
I've tried to compile with 

      BUILD_EXTRA_FLAGS=-Wextra -Wall 

and then with

    CPP_EXTRA = -Wextra -Wall 

In both cases, the flags are properly inserted into the "command line", what change is the position. I think the problem is the initial **-w** flag, that should suppress any warning...
Here an example with CPP_EXTRA:

> "C:/Users/fab/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include" "-IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include" "-IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:/msys64/tmp/mkESP/temp_sensor_d1_mini/core"  -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"  -DFLASHMODE_DIO  -DESP8266 '-DGIT_VERSION="0.9.6-1-gdc9b2c7-dirty"' -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266 -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/variants/d1_mini -IC:/msys64/tmp/mkESP/temp_sensor_d1_mini -I./ -I./src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/DNSServer/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266HTTPClient/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266SdFat/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266SdFat/src/FatLib/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266SdFat/src/SdCard/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266SdFat/src/SpiDriver/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266SdFat/src/SpiDriver/boards/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/examples/FSBrowser/extras/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/detail/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/uri/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/include/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/SD/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/SDFS/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/SPI/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/Ticker/src/ -IC:/Users/fab/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/Wire/ -IC:/Users/fab/workspace_arduino/libraries/Adafruit_BMP085_Library/ -IC:/Users/fab/workspace_arduino/libraries/Adafruit_BMP280_Library/ -IC:/Users/fab/workspace_arduino/libraries/Adafruit_NeoPixel/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/Data/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/Serialization/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/StringTraits/ -IC:/Users/fab/workspace_arduino/libraries/ArduinoJson/src/ArduinoJson/TypeTraits/ -IC:/Users/fab/workspace_arduino/libraries/RTClib-adafruit/src/ -IC:/Users/fab/workspace_arduino/libraries/WEMOS_SHT3x_by_fabz/src/ -IC:/Users/fab/workspace_arduino/libraries/bmp085decorator_lib/src/ -IC:/Users/fab/workspace_arduino/libraries/castable_lib/src/ -IC:/Users/fab/workspace_arduino/libraries/esp_logger_lib/src/ -IC:/Users/fab/workspace_arduino/libraries/rtc_memory_lib/src/ -IC:/Users/fab/workspace_arduino/libraries/settings_variable_lib/src/ -IC:/Users/fab/workspace_arduino/libraries/staticled_lib/src/  -Wextra -Wall /c/Users/fab/workspace_arduino/libraries/RTClib-adafruit/src/RTClib.cpp -o C:/msys64/tmp/mkESP/temp_sensor_d1_mini/RTClib.cpp.o

plerup, 2020-05-20:
Ok, true.
Not documented (yet) but the warning options can actually be controlled via the variable: COMP_WARNINGS

Its default value is -w but you can override it
COMP_WARNINGS = -Wextra -Wall

fabiuz7, 2020-05-20:
Top! Tested and it works, thanks

closed 2020-05-20 by fabiuz7