plerup/makeEspArduino #138: set_git_version behaviour

fabiuz7, 2020-05-21:
Hi,
a question about _set_git_version_. If I don't have that version on my PC, does it download the specified version? If yes, where is it downloaded? Could the path be personalized?

plerup, 2020-05-23:
This function assumes that you are using a cloned git repo located in a directory specified via ESP_ROOT.
Any version available in that repo can be used but you need to keep it updated for newer ones by doing a "git pull" when required.

fabiuz7, 2020-05-24:
So basically that command will run something like "git checkout <tag>", right?

plerup, 2020-05-24:
Well, that and make sure that all possible remaining files from the previous version is removed, that all sub-modules are updated and finally running the tools get command to retrieve all necessary things that are not in the repo

fabiuz7, 2020-05-24:
That's great, thanks!

closed 2020-05-24 by fabiuz7