plerup/makeEspArduino #139: git version is always dirty

fabiuz7, 2020-05-24:
Sorry if I bother you with another issue, but this time I found that even if my working directory is perfectly clean when I run *make*, _buildInfo always shows the dirty flag.

For example:

    _tBuildInfo _BuildInfo = {"2020-05-24","16:00:41","0.9.7-dirty","2.7.1"};

If I check with "git status", everything is clear, even after compilation. Is it possible that the makefile modifies or creates some temporary files when it is compiling?

Also, the final report shows the dirty flag:

> Creating object archive
Linking C:/msys64/tmp/mkESP/temp_sensor_d1_mini/temp_sensor.bin
  Versions: 0.9.7-dirty, 2.7.1
Creating BIN file "C:/msys64/tmp/mkESP/temp_sensor_d1_mini/temp_sensor.bin" using "C:/msys64/tmp/mkESP/temp_sensor_d1_mini/temp_sensor.elf"
Memory usage
  RAM:    46252 bytes
  Flash: 392384 bytes
LwIPVariant: v2 Lower Memory
Flash size: 4MB (FS:3MB OTA:~512KB)
Build complete. Elapsed time: 26 seconds

plerup, 2020-05-24:
Nothing is created outside the build root directory but if you have that underneath the source repo root directory this could happen. But then you can see it by issuing a "git status" after the build.
The build-info string is obtained from running "git describe --tags --dirty"

fabiuz7, 2020-05-25:
Finally I got the issue: I'm working on Windows, but I'm running makefile (and consequently git) from MSYS2 terminal. I had 2 different versions of git, differently configured: one native window, and the other installed in MSYS2. The working tree was dirty because I usually commit Unix style (just LF line ending) but the files on disk are windows style (CRLF).

closed 2020-05-25 by fabiuz7