plerup/makeEspArduino #141: Azure IoT - Previous Declaration

brianorbrain, 2020-08-26:
I am trying to get the Azure IoT hub to compile via makeEspArduino but not having any luck with a previous declaration.

To start, I can compile the example file inside of https://github.com/Azure/azure-iot-arduino/tree/master/examples/esp8266/iothub_ll_telemetry_sample

A few changed I made was to comment out the round, and changed the Azure Platform Local to build.extra_flags= -DDONT_USE_UPLOADTOBLOB -DUSE_BALTIMORE_CERT and it will compile perfectly in the Arduino IDE with the following output.

[arduino_output.txt](https://github.com/plerup/makeEspArduino/files/5133102/arduino_output.txt)

Now I tried a minimum viable test with just the 3 files plus the makeEspArduino and I start running into compile issues.

<details>
[brain@Craptop-Deux ino]$ make -f makeEspArduino.mk
core_esp8266_wiring.cpp
In file included from /home/brain/Arduino/libraries/AzureIoTUtility/src/umock_c/aux_inc/cstdint:7:0,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/char_traits.h:376,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/string:40,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/random:41,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Arduino.h:238,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/wiring_private.h:31,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp:22:
/home/brain/Arduino/libraries/AzureIoTUtility/src/umock_c/aux_inc/stdint.h:25:28: error: conflicting declaration 'typedef signed char int_fast8_t'
 typedef signed char        int_fast8_t;
                            ^
In file included from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:6:0,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/wiring_private.h:28,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp:22:
/home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/stdint.h:74:30: error: 'int_fast8_t' has a previous declaration as 'typedef int int_fast8_t'
   typedef __INT_FAST8_TYPE__ int_fast8_t;
                              ^
In file included from /home/brain/Arduino/libraries/AzureIoTUtility/src/umock_c/aux_inc/cstdint:7:0,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/char_traits.h:376,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/string:40,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/random:41,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:65,
                 from /home/brain/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/Arduino.h:238,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/wiring_private.h:31,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp:22:
/home/brain/Arduino/libraries/AzureIoTUtility/src/umock_c/aux_inc/stdint.h:29:28: error: conflicting declaration 'typedef unsigned char uint_fast8_t'
 typedef unsigned char      uint_fast8_t;
                            ^
In file included from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:6:0,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/wiring_private.h:28,
                 from /home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp:22:
/home/brain/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sdk/libc/xtensa-lx106-elf/include/stdint.h:75:31: error: 'uint_fast8_t' has a previous declaration as 'typedef unsigned int uint_fast8_t'
   typedef __UINT_FAST8_TYPE__ uint_fast8_t;
</details>

The arduino.mk for this process was :
<details>
# Board definitions
CORE_DEBUG_LEVEL ?= 
F_CPU ?= 80000000L
FLASH_MODE ?= qio
FLASH_SPEED ?= 40
UPLOAD_RESET ?= --before default_reset --after hard_reset
UPLOAD_SPEED ?= 115200
COMP_WARNINGS ?= -w
INCLUDE_VARIANT = nodemcu
# Commands
C_COM=$(C_COM_PREFIX) "$(ESP_ROOT)/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I$(ESP_ROOT)/tools/sdk/include" "-I$(ESP_ROOT)/tools/sdk/lwip2/include" "-I$(ESP_ROOT)/tools/sdk/libc/xtensa-lx106-elf/include" "-I$(BUILD_DIR)/core" $(C_PRE_PROC_FLAGS) -c $(COMP_WARNINGS) -std=c17  -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DNONOSDK22x_190703=1 -DF_CPU=$(F_CPU) -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_$(CHIP) -DARDUINO_BOARD=\"ESP8266_NODEMCU\"  -DFLASHMODE_QIO  -DESP8266 $(BUILD_EXTRA_FLAGS) $(C_INCLUDES) 
CPP_COM=$(CPP_COM_PREFIX) "$(ESP_ROOT)/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I$(ESP_ROOT)/tools/sdk/include" "-I$(ESP_ROOT)/tools/sdk/lwip2/include" "-I$(ESP_ROOT)/tools/sdk/libc/xtensa-lx106-elf/include" "-I$(BUILD_DIR)/core" $(C_PRE_PROC_FLAGS) -c $(COMP_WARNINGS)  -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DNONOSDK22x_190703=1 -DF_CPU=$(F_CPU) -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_$(CHIP) -DARDUINO_BOARD=\"ESP8266_NODEMCU\"  -DFLASHMODE_QIO  -DESP8266 $(BUILD_EXTRA_FLAGS) $(C_INCLUDES) 
S_COM="$(ESP_ROOT)/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I$(ESP_ROOT)/tools/sdk/include" "-I$(ESP_ROOT)/tools/sdk/lwip2/include" "-I$(ESP_ROOT)/tools/sdk/libc/xtensa-lx106-elf/include" "-I$(BUILD_DIR)/core" $(C_PRE_PROC_FLAGS) -c -g -x assembler-with-cpp -MMD -mlongcalls -DNONOSDK22x_190703=1 -DF_CPU=$(F_CPU) -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_$(CHIP) -DARDUINO_BOARD=\"ESP8266_NODEMCU\"  -DFLASHMODE_QIO  -DESP8266 $(BUILD_EXTRA_FLAGS) $(C_INCLUDES) 
LIB_COM="$(ESP_ROOT)/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar"
[... 55 lines cut ...]
In file included from /home/brain/ino_test/lib/packages/AzureIoTUtility/src/umock_c/aux_inc/cstdint:7,
                 from /home/brain/ino_test/lib/Arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/memory:94,
                 from /home/brain/ino_test/lib/Arduino/cores/esp8266/FS.h:24,
[... cut ...]
/home/brain/ino_test/lib/Arduino/tools/sdk/libc/xtensa-lx106-elf/include/stdint.h:75:31: note: previous declaration as 'typedef unsigned int uint_fast8_t'
   75 |   typedef __UINT_FAST8_TYPE__ uint_fast8_t;
      |                               ^~~~~~~~~~~~
</details>

At this point, I am lost as to what is causing the makefile to handle the Azure IoT files differently as from the Arduino IDE.

If you need any other info, please let me know as this is bugging me why it works one place and not the other.

Thanks
Brian

plerup, 2020-11-27:
I haven't had time to look into this, sorry, but please check using the latest commit. The order of include directories have changed

plerup, 2021-11-18:
Closing due to inactivity

closed 2021-11-18 by plerup