plerup/makeEspArduino #144: Exception 29 (StoreProhibited) on LittleFS.openDir("a")

claudiomattera, 2020-11-12:
Hello everyone,

I want to use LittleFS to work with directories, but I get an exception 29 (StoreProhibitedCause) as soon as I try to open one.

I opened an [issue on esp8266/Arduino](https://github.com/esp8266/Arduino/issues/7683) repository, but the problem could not be replicated, and it was suggested it might be related to the makefile itself, so I am posting it here as well.

I use the commands:

~~~~
espmake erase_flash
espmake flash_fs
espmake run
~~~~

I tried this sketch on two different NodeMCUv2 units from two different vendors and I get the same results.
No wires are attached to any of the pins.

I tried core version 2.7.4 and 2.6.0, and the latest commit available on master, and I get the same results.
I also tried to install the makefile and the toolchain from scratch on a brand new Raspberry Pi, and to compile and flash my sketch from there, but the results are the same.

However, I also tried to compile and flash the sketch using a different toolchain (an old [Esp8266-Arduino-Makefile](https://github.com/thunderace/Esp8266-Arduino-Makefile) I used to use), and it works perfectly.
So it does not look like a hardware problem (using that toolchain I can also run more complex sketches where I read and write files in a directory).

### Sketch

```cpp
#include <Arduino.h>
#include <LittleFS.h>

void setup(void)
{
    Serial.begin(115200);

    LittleFS.begin();
    Serial.println("Formatting");
    Serial.println(LittleFS.format());
    Serial.println("Formatted");

    Serial.println("Creating dir");
    Serial.println(LittleFS.mkdir("a"));
    Serial.println("Created dir");

    Serial.println("Opening dir"); // <- This is the last output I get
    Dir directory = LittleFS.openDir("a");
    Serial.println("Opened dir");

    LittleFS.end();
}

void loop(void)
{
}
```

~~~~
# File config.mk
THIS_DIR := $(realpath $(dir $(realpath $(lastword $(MAKEFILE_LIST)))))
ROOT := $(THIS_DIR)/..
LIBS = $(ESP_LIBS)/LittleFS

SKETCH = $(THIS_DIR)/dir-test.cpp

UPLOAD_PORT = /dev/ttyUSB0
UPLOAD_SPEED = 115200

# Enable high-performance network stack with IPv6 support
LWIP_VARIANT = hb6f

FS_TYPE = littlefs

BUILD_EXTRA_FLAGS = -DARDUINO_ARCH_ESP8266
~~~~

~~~~
# Makefile for Esp8266-Arduino-Makefile
ARDUINO_VARIANT = nodemcuv2
SERIAL_PORT = /dev/ttyUSB0
SERIAL_BAUD = 115200
LOG_SERIAL_TO_FILE=yes

include ../Esp8266-Arduino-Makefile/espXArduino.mk
~~~~

### Debug Messages

```
Formatting
1
Formatted
Creating dir
1
Created dir
Opening dir

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x40201d0b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000020 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc80 end: 3fffffc0 offset: 0190
3ffffe10:  3ffefc84 00000400 3ffffedc 40201c84
3ffffe20:  00000008 3ffeff94 00000001 00000040
3ffffe30:  3ffeff95 00000000 3ffffe90 40201eac
3ffffe40:  3ffeff94 00000000 3ffffe90 00200401
3ffffe50:  3ffffe00 00000000 00000004 00000000
3ffffe60:  3fffff78 3ffffedc 3ffefc84 00000000
3ffffe70:  00000008 3ffefc84 00000000 40202b6c
3ffffe80:  500ffc1b 00010002 00000078 00000079
3ffffe90:  3ffeff94 402078c0 00000040 40206687
3ffffea0:  3fffff78 3ffefc84 3ffeff94 40207912
3ffffeb0:  00000004 00000080 500ffc1b 3ffee9f8
3ffffec0:  402088ec 3fffff1f 3ffee9f8 402088f8
3ffffed0:  402088ec 3fffff1f 3ffee9f8 40208002
3ffffee0:  00000001 00000061 00000000 00000000
3ffffef0:  00000000 00000000 00000000 00000000
3fffff00:  00000000 00000000 00000000 00000000
3fffff10:  00000000 00000000 00000000 00000000
3fffff20:  00000000 00000000 00000000 00000000
3fffff30:  00000000 00000000 00000000 00000000
3fffff40:  00000000 00000000 00000000 00000000
3fffff50:  00000000 00000000 00000000 00000000
3fffff60:  00000000 00000000 00000000 00000000
3fffff70:  00000000 00000000 00000000 00000000
3fffff80:  00000000 00000000 00000000 00000000
3fffff90:  00000000 00000000 00000000 00000000
3fffffa0:  00000000 00000000 00000000 00000000
3fffffb0:  00000000 00000000 00000000 00000000
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3640, room 16
```

If I decode the stack trace with `espmake crash` I get this:

~~~~
Exception (29):
StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores

=== Stack trace ===

lfs_dir_find
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1102
lfs_dir_get
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:582
lfs_dir_getinfo
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1048
lfs_dir_open
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:2011
__gnu_cxx::new_allocator<std::_Sp_counted_ptr_inplace<lfs_dir, std::allocator<lfs_dir>, (__gnu_cxx::_Lock_policy)0> >::allocate(unsigned int, void const*)
  /home/claudio/projects/micro-controllers/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.2.0/ext/new_allocator.h:115
operator new(unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/abi.cpp:41
littlefs_impl::LittleFSImpl::openDir(char const*)
  /home/claudio/projects/micro-controllers/esp8266/libraries/LittleFS/src/LittleFS.cpp:122
HardwareSerial::write(unsigned char const*, unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/HardwareSerial.h:164
HardwareSerial::write(unsigned char const*, unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/HardwareSerial.h:165
HardwareSerial::write(unsigned char const*, unsigned int)
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/HardwareSerial.h:164
settimeofday
  /home/claudio/projects/micro-controllers/esp8266/cores/esp8266/time.cpp:244
~~~~

### Platform

- Hardware:             [ESP-12]
- Core Version:         [77bd71ec2987849491ec5d6d889eb77bad882f67 (latest commit on 2020-10-29)]
- Development Env:      [Make]
- Operating System:     [Linux]

### Settings in IDE

- Module:           [Nodemcu v2]
- Flash Mode:       [?]
- Flash Size:       [4MB]
- lwip Variant:     [v2 Higher Bandwidth]
- Reset Method:     [?]
- Flash Frequency:  [?]
- CPU Frequency:    [?]
- Upload Using:     [SERIAL]
- Upload Speed:     [115200]

Note: I am not entirely sure about some of these fields, I filled them up to the best of my knowledge.

plerup, 2020-11-12:
Remove the line:
LIBS = $(ESP_LIBS)/LittleFS

claudiomattera, 2020-11-13:
Removing that line does indeed solve the problem, thanks.

In my actual sketch I had more libraries listed in `LIBS`, and if I simply removed LittleFS I got a linker error for symbol LittleFS not found.
I solved that by ditching `LIBS` entirely and by setting `CUSTOM_LIBS` to the directories of my additional libraries (the ones not bundled with the toolchain).

The only drawback (if I can even call it that) is that now I must explicitly `#include` each library in the main source file, even those that are only used by other libraries, otherwise it will not be linked.

closed 2020-11-13 by claudiomattera