plerup/makeEspArduino #145: Link time wrapping functions is broken since implementing user_obj.ar trick.

everslick, 2020-11-15:
I need to wrap some functions to trace memory allocation like:

```
LD_EXTRA = -Wl,-wrap,malloc      \
                  -Wl,-wrap,calloc      \
                  -Wl,-wrap,realloc     \
                  -Wl,-wrap,free
```

This used to work fine even though the espressif SDK does the same. Since the `user_obj.ar` trick was introduced this breaks.

```
Creating core archive
Linking ../build/3.0.99/efate/eng/alpha/emonio.bin
  Versions: 3.0.29-70-gbbbc570d-dirty, 1.0.5-rc1-21-gcee7b423
../build/3.0.99/efate/eng/alpha/user_obj.ar(wrapper.cpp.o): In function `__wrap_malloc':
/home/clemens/Devel/ESP/emonio-fw/src/wrapper.cpp:99: multiple definition of `__wrap_malloc'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_trace.c:395: first defined here
../build/3.0.99/efate/eng/alpha/user_obj.ar(wrapper.cpp.o): In function `__wrap_calloc':
wrapper.cpp:(.text.__wrap_calloc+0x0): multiple definition of `__wrap_calloc'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):heap_trace.c:(.iram1.25+0x0): first defined here
../build/3.0.99/efate/eng/alpha/user_obj.ar(wrapper.cpp.o): In function `__wrap_realloc':
wrapper.cpp:(.text.__wrap_realloc+0x0): multiple definition of `__wrap_realloc'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):heap_trace.c:(.iram1.24+0x0): first defined here
../build/3.0.99/efate/eng/alpha/user_obj.ar(wrapper.cpp.o): In function `__wrap_free':
wrapper.cpp:(.text.__wrap_free+0x0): multiple definition of `__wrap_free'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):heap_trace.c:(.iram1.23+0x0): first defined here
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):(.iram1.19.literal+0x0): undefined reference to `__real_heap_caps_malloc'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):(.iram1.19.literal+0x4): undefined reference to `__real_heap_caps_malloc_default'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):(.iram1.20.literal+0x0): undefined reference to `__real_heap_caps_free'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):(.iram1.21.literal+0x0): undefined reference to `__real_heap_caps_realloc'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o):(.iram1.21.literal+0x4): undefined reference to `__real_heap_caps_realloc_default'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o): In function `trace_malloc':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_trace.c:145: undefined reference to `__real_heap_caps_malloc'
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_trace.c:145: undefined reference to `__real_heap_caps_malloc_default'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o): In function `trace_free':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_trace.c:145: undefined reference to `__real_heap_caps_free'
../../Arduino-ESP32-espressif/tools/sdk/lib/libheap.a(heap_trace.o): In function `trace_realloc':
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_trace.c:145: undefined reference to `__real_heap_caps_realloc'
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_trace.c:145: undefined reference to `__real_heap_caps_realloc_default'
collect2: error: ld returned 1 exit status
make[1]: *** [../../makeEspArduino/makeEspArduino.mk:335: ../build/3.0.99/efate/eng/alpha/emonio.bin] Error 1
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/src'
make: *** [Makefile:46: src] Error 2
```

Does anyone have an idea for a workaround to make it link again? I think the order in which archives and/or object files are linked might be relevant or even the order of command line parameters. Or could we have a switch to make the `ar` trick optional? 

makeEspArduino is a wonderful tool, I love it! Thanks for making it!

plerup, 2020-11-15:
We could do something like this, on line 328

USER_OBJ_DEP ?= $(USER_OBJ_LIB)
$(MAIN_EXE): $(CORE_LIB) $(USER_LIBS) $(USER_OBJ_DEP)

And then you could use this on the command line to avoid putting the object files into the libarary

USER_OBJ_DEP="\$(USER_OBJ)"

Can you check if this solves your problem?

everslick, 2020-11-15:
Thanks so much for your input. Based on your idea I created a [PR](https://github.com/plerup/makeEspArduino/pull/146) for what worked for me.

everslick, 2020-11-15:
Should you decide to merge the PR, I will happily create another PR to update the documentation as well.

everslick, 2020-11-15:
Sent 2 new PRs. Closing this issue! Thank you very much!

closed 2020-11-15 by everslick