plerup/makeEspArduino #149: Failing build when catch.hpp is present

krenzlin, 2020-11-20:
I have a weird behaviour when the header file of catch2 is in the root or sub-dir of the sketch.
Building fails with a complete unrelated (?) error

```
Ethernet.cpp
In file included from /home/konrad/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/Ethernet/src/Ethernet.cpp:1:0:
/home/konrad/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/Ethernet/src/utility/w5100.h:13:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>
                 ^
compilation terminated.
```

**Reproduce**

* clone `makeEspArduino`
* create minimal sketch `main.ino`
```
void setup() {};
void loop() {};
```
* run `make -f makeEspArduino.mk SKETCH=main.ino`
  * correctly builds
* get catch2 header `wget https://raw.githubusercontent.com/catchorg/Catch2/v2.x/single_include/catch2/catch.hpp`
* run `make -f makeEspArduino.mk SKETCH=main.ino`
  * fails with above error

Also putting the header into a sub-dir (e.g. tests/) and using `EXCLUDE_DIRS=./tests` doesn't work.
I even tried putting the whole header in a `#ifndef ARDUINO`

Any ideas would help

plerup, 2020-11-22:
Yes, confirmed as an error. The search gets hold of a directory which should not be used

However, a complete new include search algorithm is already in the making and a new version of makeEspArduino is expected to be ready within a week

krenzlin, 2020-11-22:
Looking forward to that. By the way thanks for your great work!

plerup, 2020-11-27:
Please check the latest commit

krenzlin, 2020-11-27:
Thanks! It works. 
All the best

closed 2020-11-27 by krenzlin