plerup/makeEspArduino #150: Building against IDF4.2 branch of Arduino Core

everslick, 2020-11-23:
Sooner or later the IDF4.2 branch will be merged into master, thats why I gave it a shot already.

Switching to IDF4.2:

```
cd arduino-esp32
git switch idf-release/v4.2
cd tools
./get.py
```

There are now two toolchains: **xtensa-esp32-elf** and **xtensa-esp32s2-elf**
And two SDK directories: **esp32** and **esp32s2**

Build seems to use **esp32s2** by default.

`ccache "../../Arduino-ESP32-espressif/tools/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++" -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE -DIDF_VER=\"v4.3-dev-1720-g494a124d9-dirty\" -DESP_PLATFORM  ...`

Maybe we need a new CHIP=esp32s2 ?

plerup, 2020-11-24:
Ok, will check
Big new update is in the works anyway :fearful: 
Brace yourself :wink:

everslick, 2020-11-24:
Can't wait to give it a shot! :-D

plerup, 2020-11-24:
esp32s2 is the first board in the list and hence becomes the default
use BOARD=esp32 to get the generic esp32 instead

everslick, 2020-11-25:
damn! never set BOARD because the default matched anyway, thanks!

closed 2020-11-25 by everslick