plerup/makeEspArduino #151: no installed version of esp8266

zonemikel, 2020-11-25:
make -f makeEspArduino.mk DEMO=1 flash
makeEspArduino.mk:112: *** No installed version of esp8266 Arduino found.  Stop.

I've installed ardurino ide and 8266 through the board manager

plerup, 2020-11-25:
Which OS is this?
Does the directory ~/.arduino15/packages/esp8266 exist?

zonemikel, 2020-11-25:
Thank you for responding
Linux/ubuntu 
Linux Mint 20 Ulyana \n \l
Yes it does, here is some of the files, is there a specific file it seeks ? 
I'm just trying to edit the ino files in vim and then run a command that compiles and pushes to the esp8266. I can do it fine within the arduino ide.

What env vars could I define before running to make it work ? 

`/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/examples/NTPClient
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/examples/IPv6
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/examples/IPv6/IPv6.ino
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/examples/WiFiClient
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/package.json
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/keywords.txt
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/programmers.txt
/home/michael/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/boards.txt
`

plerup, 2020-11-25:
Try using:

make -f makeEspArduino.mk DEMO=1 ARDUINO_ROOT=/home/michael/.arduino15

zonemikel, 2020-11-26:
That worked ! It built the file, not sure why it didn't work before without the ARDRUINO_ROOT I'll move further questions into the wiki as they aren't issues. Let me know if you need anything more about my setup I can provide.

plerup, 2020-11-26:
Please check your HOME environment variable

zonemikel, 2020-11-26:
echo $HOME
/home/michael

plerup, 2020-11-27:
Try putting this into a makefile and then use it as input for make

all:
	echo $(shell uname -s)
	echo $(HOME)
	ls -l $(HOME)/.arduino15

plerup, 2021-11-18:
Closing due to inactivity

closed 2021-11-18 by plerup