plerup/makeEspArduino #153: Cannot open linker script file local.eagle.flash.ld

schlan, 2020-12-13:
Hey, I'm trying to build the sample code via 

```
make -f makeEspArduino.mk DEMO=0 all ESP_ROOT=/home/seb/Arduino/hardware/esp8266com/esp8266
```

and I'm running into this error:

```
Creating core archive
Creating object archive
Linking /tmp/mkESP/WiFiScan_generic/WiFiScan.bin
  Versions: 2.7.2-188-g1e016a42, 2.7.2-188-g1e016a42
/home/seb/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file local.eagle.flash.ld: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [makeEspArduino.mk:326: /tmp/mkESP/WiFiScan_generic/WiFiScan.bin] Error 1
```

plerup, 2020-12-13:
Please do a run using VERBOSE=1

plerup, 2021-01-25:
Fixed in latest commit

closed 2021-01-25 by plerup