plerup/makeEspArduino #155: Can't compile since 8df08963

spacelama, 2021-01-29:
With the "recursive compile everything" of 8df08963, it is no longer possible to compile anything if your personal arduino libraries include a library with test functions that don't compile on your system:

````
"/home/tconnors/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/include" "-I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/lwip2/include" "-I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/mkESP/ac-controller_d1_mini/core"  -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"  -DFLASHMODE_DIO  -DESP8266 -D_SSID=\"router2.4\" -D_SSID2=\"re4502.4\" -D_PWD=\"EnrotalphEg4ChrebfeOgEr1\" -D_SYSLOG=\"192.168.1.17\" -D_OTA_PWD=\"4quitVojwigig\" -D_HOSTNAME=\"ac\" -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266 -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/variants/d1_mini -I/tmp/mkESP/ac-controller_d1_mini -I/home/tconnors/Arduino/ac-controller -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WiFi/src/include -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ArduinoOTA -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266mDNS/src -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src -I/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266WebServer/src/detail -I/home/tconnors/Arduino/libraries/Arduino_JSON/src -I/home/tconnors/Arduino/libraries/Arduino_JSON/src/cjson -I/home/tconnors/Arduino/libraries/AsyncPing/src -I/home/tconnors/Arduino/libraries/HeatpumpIR -I/home/tconnors/Arduino/libraries/ArduinoJson/extras/tests/Helpers -I/home/tconnors/Arduino/libraries/ArduinoJson/extras/tests/Helpers/api -I/home/tconnors/Arduino/libraries/ESP_EEPROM/src -I/home/tconnors/Arduino/libraries/IRremoteESP8266/test -I/home/tconnors/Arduino/libraries/IRremoteESP8266/src -I/home/tconnors/Arduino/libraries/IRremoteESP8266/src/locale   /home/tconnors/Arduino/libraries/IRremoteESP8266/test/ir_Argo_test.cpp -o /tmp/mkESP/ac-controller_d1_mini/ir_Argo_test.cpp.o
/home/tconnors/Arduino/libraries/IRremoteESP8266/test/IRac_test.cpp:38:25: fatal error: gtest/gtest.h: No such file or directory
 #include "gtest/gtest.h"
````

That git commit talks about it might break certain setups with certain $LIBS, but I haven't set LIBS in any way.  I preferred the old way of scanning #includes, since that worked for all the code I was compiling.

plerup, 2021-01-29:
Can you supply an example?

spacelama, 2021-02-09:
Hopefully I've captured all the dependencies in http://rather.puzzling.org/~tconnors/tmp/makeEspAruino-155.tgz (well, all except for Arduino libraries it depends on in ~/Arduino/libraries )

Extract to a temporary fresh copy of ~/Arduino (because some paths are hardcoded, but will overwrite makeEspArduino).  Make sure the arduino IDE has ESP8266WiFi WiFiClient ESP8266WebServer ESP8266mDNS WiFiUdp ArduinoOTA HeatpumpIP (and probably others).  Perhaps make sure your computer doesn't have any gtest libraries installed, although my debian machine does have libgtest-dev.

Run `make` in ac-controller dir - the OTA flash/WPA credentials in ~/Arduino/template won't need to be changed for the test to see whether this compiles.

The version of makeEspArduino that I've checked out in that directory ought work, but if you checkout the next commit (8df08963) instead, then it fails, because of the aforementioned gtest file in IRremoteESP8266/test/IRac_test.cpp.

plerup, 2021-02-09:
Try adding this definition

    EXCLUDE_DIRS=IRremoteESP8266/test/

spacelama, 2021-02-09:
On Tue, 9 Feb 2021, Peter Lerup wrote:

> Try adding this definition
>
>     EXCLUDE_DIRS=IRremoteESP8266/test/

That might get me a little further, but it quickly becomes intractible
pulling in plenty of other uneeded dependencies, before I get stuck:

EXCLUDE_DIRS=$(wildcard $(ARDUINO_LIBS)/*/tests) $(ARDUINO_LIBS)/ESP8266SdFat $(ARDUINO_LIBS)/SDFS

0-0-22:27:04, Tue Feb 09 tconnors@dirac:~/Arduino/ac-controller [master|✚ 2] (bash)
65893,91> make
istream.cpp
Adafruit_GrayOLED.cpp
Adafruit_SPITFT.cpp
glcdfont.c
Adafruit_ZeroDMA.cpp
JSON.cpp
JSONVar.cpp
cJSON.c
AsyncPing.cpp
/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266SdFat/src/FatLib/istream.cpp:
In member function 'bool sdfat::istream::getDouble(double*)':
/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266SdFat/src/FatLib/istream.cpp:204:17:
error: 'FLT_MIN' was not declared in this scope
         if (v < FLT_MIN * pow10  && frac != 0) {
                 ^
/home/tconnors/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266SdFat/src/FatLib/istream.cpp:210:17:
error: 'FLT_MAX' was not declared in this scope
         if (v > FLT_MAX / pow10) {
                 ^
make: *** [/home/tconnors/Arduino/makeEspArduino/makeEspArduino.mk:272:
/tmp/mkESP/ac-controller_d1_mini/istream.cpp.o] Error 1

-- 
Tim Connors

plerup, 2021-02-09:
Not quite sure what's happening here but ESP8266SdFat which generates the error i not in $(ARDUINO_LIBS)/ESP8266SdFat  but in $(ESP_LIBS)/ESP8266SdFat 
Don't know why you needed to exclude it though
Investigation would be very much easier if you could just pack your complete ~/Arduino and ~/.arduino15 in an archive and provide it

everslick, 2021-08-14:
I also have problems with the new LIBS collector. When a library contains C and C++ files, the C files don't get picked up for compilation and linkage. This happens for example with the NimBLE bluetooth stack on ESP32.

plerup, 2021-08-17:
This is because the source files are located in another directory than the corresponding header file.
I will try and make a solution for this

plerup, 2021-08-18:
The coming version will have a new variable named EXPAND_LIBS to remedy this problem.
When set all the source files found in the directories specified by LIBS will be added to the build, i.e. same behavior as before version 6.0

plerup, 2021-08-18:
Available now

everslick, 2021-09-16:
maybe related, but not sure about it:

with current Arduino CORE (2.0.0) linking fails, because it finds the same symbol in **arduino.ar** and **main.cpp**

```
/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../build/3.0.99/jarvis/eng/alpha/main.cpp.o:/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:34: multiple definition of `loopTaskWDTEnabled'; ../build/3.0.99/jarvis/eng/alpha/arduino.ar(main.cpp.o):/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:34: first defined here
/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../build/3.0.99/jarvis/eng/alpha/main.cpp.o: in function `loopTask(void*)':
/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:37: multiple definition of `loopTask(void*)'; ../build/3.0.99/jarvis/eng/alpha/arduino.ar(main.cpp.o):/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:37: first defined here
/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../build/3.0.99/jarvis/eng/alpha/main.cpp.o:/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:20: multiple definition of `loopTaskHandle'; ../build/3.0.99/jarvis/eng/alpha/arduino.ar(main.cpp.o):/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:20: first defined here
/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../build/3.0.99/jarvis/eng/alpha/main.cpp.o: in function `app_main':
/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:52: multiple definition of `app_main'; ../build/3.0.99/jarvis/eng/alpha/arduino.ar(main.cpp.o):/home/clemens/Devel/ESP/Arduino-ESP32-2.0.0/cores/esp32/main.cpp:52: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [../../makeEspArduino/makeEspArduino.mk:340: ../build/3.0.99/jarvis/eng/alpha/emonio.bin] Error 1
make[1]: *** [Makefile:70: src] Error 2
```

everslick, 2021-09-16:
but I can confirm that `EXPAND_LIBS=1` fixes my issue with **NimBLE**.

plerup, 2021-09-16:
The DEMO option works for 2.0.0 so I can't reproduce this

everslick, 2021-09-17:
I will get back to this issue later, thanks for checking.

closed 2022-06-08 by plerup