plerup/makeEspArduino #157: Why are you using perl ?

Animeshdhakal, 2021-02-04:
Why are you using perl instead of python. I am curious about it

plerup, 2021-02-05:
Because I have been using Perl for many many years and think I know it well :older_man: 
I do use Python as well but when it comes to writing compact code for efficient text file processing I prefer Perl.
The original versions of makeEspArduino actually used the Perl code inline in the makefile in order to keep the file footprint low but as the scripts became more advanced this made it hard to maintain. So I extracted them out to separate files and suddenly the main language of this repo became Perl according to Github :grin:

Animeshdhakal, 2021-02-05:
Can I rewrite this in python. Where are the docs for the core of Esp8266 Arduino

plerup, 2021-02-05:
Why?
You can of course do whatever you want, the code is open. I will not change this in the repo though
There is no documentation, my findings are done the hard way

closed 2021-02-06 by Animeshdhakal