plerup/makeEspArduino #160: Compiling Error

Animeshdhakal, 2021-02-21:
It give me this error
ArduinoOTA.h: No such file or directory
 #include <ArduinoOTA.h>

plerup, 2021-02-21:
Well hard to say without any information about your OS or setup

An empty sketch like this works on Linux for sure

    #include <ArduinoOTA.h> 
    void setup() {}
    void loop() {}

Animeshdhakal, 2021-02-22:
I am using Kali Linux 

and this is my makefile

ESP_ROOT = ./build/esp8266/
SKETCH = ./src/main.cpp
LIBS = ./lib
BOARD = nodemcu

include ./build/makeEspArduino/makeEspArduino.mk

plerup, 2021-02-22:
What happens if you replace the contents of your sketch with the lines I posted above?

Also try

     make list_lib

You should also verify the the file:

    ./build/esp8266//libraries/ArduinoOTA/ArduinoOTA.h

really exists

Animeshdhakal, 2021-02-22:
Yep It was my fault

closed 2021-02-22 by Animeshdhakal