plerup/makeEspArduino #161: When I use master version it gives me littlefs error

Animeshdhakal, 2021-03-14:
In file included from /home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:8:
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c: In function 'lfs_alloc':
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:494:45: error: expected ')' before '__INT32'
  494 |             LFS_ERROR("No more free space %"PRIu32,
      |                                             ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:78:28: note: in definition of macro 'LFS_ERROR_'
   78 |     printf("%s:%d:error: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:494:13: note: in expansion of macro 'LFS_ERROR'
  494 |             LFS_ERROR("No more free space %"PRIu32,
      |             ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c: In function 'lfs_dir_fetchmatch':
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:997:43: error: expected ')' before '__INT32'
  997 |     LFS_ERROR("Corrupted dir pair at {0x%"PRIx32", 0x%"PRIx32"}",
      |                                           ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:78:28: note: in definition of macro 'LFS_ERROR_'
   78 |     printf("%s:%d:error: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:997:5: note: in expansion of macro 'LFS_ERROR'
  997 |     LFS_ERROR("Corrupted dir pair at {0x%"PRIx32", 0x%"PRIx32"}",
      |     ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c: In function 'lfs_dir_compact':
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1536:58: error: expected ')' before '__INT32'
 1536 |                 LFS_DEBUG("Expanding superblock at rev %"PRIu32, dir->rev);
      |                                                          ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:62:28: note: in definition of macro 'LFS_DEBUG_'
   62 |     printf("%s:%d:debug: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1536:17: note: in expansion of macro 'LFS_DEBUG'
 1536 |                 LFS_DEBUG("Expanding superblock at rev %"PRIu32, dir->rev);
      |                 ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1685:41: error: expected ')' before '__INT32'
 1685 |             LFS_DEBUG("Bad block at 0x%"PRIx32, dir->pair[1]);
      |                                         ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:62:28: note: in definition of macro 'LFS_DEBUG_'
   62 |     printf("%s:%d:debug: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1685:13: note: in expansion of macro 'LFS_DEBUG'
 1685 |             LFS_DEBUG("Bad block at 0x%"PRIx32, dir->pair[1]);
      |             ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1690:38: error: expected ')' before '__INT32'
 1690 |             LFS_WARN("Superblock 0x%"PRIx32" has become unwritable",
      |                                      ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:70:27: note: in definition of macro 'LFS_WARN_'
   70 |     printf("%s:%d:warn: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                           ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1690:13: note: in expansion of macro 'LFS_WARN'
 1690 |             LFS_WARN("Superblock 0x%"PRIx32" has become unwritable",
      |             ^~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1707:36: error: expected ')' before '__INT32'
 1707 |         LFS_DEBUG("Relocating {0x%"PRIx32", 0x%"PRIx32"} "
      |                                    ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:62:28: note: in definition of macro 'LFS_DEBUG_'
   62 |     printf("%s:%d:debug: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:1707:9: note: in expansion of macro 'LFS_DEBUG'
 1707 |         LFS_DEBUG("Relocating {0x%"PRIx32", 0x%"PRIx32"} "
      |         ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c: In function 'lfs_ctz_extend':
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:2331:37: error: expected ')' before '__INT32'
 2331 |         LFS_DEBUG("Bad block at 0x%"PRIx32, nblock);
      |                                     ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:62:28: note: in definition of macro 'LFS_DEBUG_'
   62 |     printf("%s:%d:debug: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:2331:9: note: in expansion of macro 'LFS_DEBUG'
 2331 |         LFS_DEBUG("Bad block at 0x%"PRIx32, nblock);
      |         ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c: In function 'lfs_file_relocate':
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:2635:37: error: expected ')' before '__INT32'
[... 89 lines cut ...]
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:4119:5: note: in expansion of macro 'LFS_DEBUG'
 4119 |     LFS_DEBUG("Fixing move {0x%"PRIx32", 0x%"PRIx32"} 0x%"PRIx16,
      |     ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c: In function 'lfs_fs_deorphan':
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:4170:47: error: expected ')' before '__INT32'
 4170 |                 LFS_DEBUG("Fixing orphan {0x%"PRIx32", 0x%"PRIx32"}",
      |                                               ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:62:28: note: in definition of macro 'LFS_DEBUG_'
   62 |     printf("%s:%d:debug: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:4170:17: note: in expansion of macro 'LFS_DEBUG'
 4170 |                 LFS_DEBUG("Fixing orphan {0x%"PRIx32", 0x%"PRIx32"}",
      |                 ^~~~~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:4192:52: error: expected ')' before '__INT32'
 4192 |                 LFS_DEBUG("Fixing half-orphan {0x%"PRIx32", 0x%"PRIx32"} "
      |                                                    ^~~~~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs_util.h:62:28: note: in definition of macro 'LFS_DEBUG_'
   62 |     printf("%s:%d:debug: " fmt "%s\n", __FILE__, __LINE__, __VA_ARGS__)
      |                            ^~~
/home/animeshdhakal/Desktop/HomeAutomation/build/esp8266/libraries/LittleFS/lib/littlefs/lfs.c:4192:17: note: in expansion of macro 'LFS_DEBUG'
 4192 |                 LFS_DEBUG("Fixing half-orphan {0x%"PRIx32", 0x%"PRIx32"} "
      |                 ^~~~~~~~~
make: *** [build/makeEspArduino/makeEspArduino.mk:284: /tmp/mkESP/main_nodemcu/lfs.c.o] Error 1
make: *** Waiting for unfinished jobs....

plerup, 2021-03-14:
LittleFS has a strange directory structure which is a mix of source and a submodule. This makes makeEspArduino catch the wrong file in the wildcard search.
For now you can avoid this by adding the following to your build:

    EXCLUDE_DIRS=LittleFS/lib

I will check for possible automatic handling of this in coming versions

Animeshdhakal, 2021-03-15:
I just fix the error 
What I was doing wrong is I was doing git checkout master although I am in the master branch 
This solved my error

closed 2021-03-15 by Animeshdhakal

TooterTutor, 2022-05-28:
Hi, when I try to upload to my ESP-WROOM-32 using LittleFS, it throws an error saying 
`.arduino15/packages/esp32/hardware/esp32/2.0.3/tools/mklittlefs: 1: Syntax error: "(" unexpected`

Does anyone know how to fix this?

plerup, 2022-05-30:
Please specify the command line which you are using