plerup/makeEspArduino #162: Usage of an environment file.

Nurrl, 2021-04-29:
Hello, 

a nice feature would be to add a way to manage secrets(like Wi-Fi **SSID** and **Password**) when using this project and push the ESP project to GitHub.
Lots of languages use `.env` files for this very purpose, so I came out with an implementation like this:

*In my `Makefile`*:
```makefile
# ...
ENV = .env
# ...
include .../makeEspArduino.mk
```

*In the `.env` file*: (note this is in the `Makefile` format, because it will be included in the Makefile, but we can maybe do it another way.)
```makefile
MY_DEFINE = "My string value"
MY_INT_DEFINE = 12
```

*In `makeEspArduino.mk`*:
```makefile
#- `.env` file loading for secret management -#
ifneq ($(strip $(ENV)),)

# Include the environment file to get acces to variables
include $(ENV)

# Identify the variables to add them to `BUILD_EXTRA_FLAGS`
DEFINES	:= $(shell awk '{print $$1;}' $(ENV))
DEFINES := $(foreach define,$(DEFINES),-D$(define)='$($(define))')

BUILD_EXTRA_FLAGS += $(DEFINES)

endif
```

So this effectively adds:
```
-DMY_DEFINE='"My string value"' -DMY_INT_DEFINE='12'
```
to the `BUILD_EXTRA_FLAGS` variable.

Should I make a Pull Request ?
Thanks.

plerup, 2021-05-02:
Can't you use the two already present optional configuration files for this, the config.mk in the SKETCH directory and the one in $HOME/.config?

Nurrl, 2021-05-02:
Hello, these optional configuration files are for `makeEspArduino` only, right ? Because my usage is to pass C/C++ `#defines` using the `-D` switch on GCC from a local configuration so that I can uses those values within the code.

plerup, 2021-05-02:
Well anything you describe above can be done in these files, just add e.g. this line

`BUILD_EXTRA_FLAGS += -DMY_DEFINE='"My string value"' -DMY_INT_DEFINE='12'`

Nurrl, 2021-05-02:
Obviously yes, thanks for your help.

closed 2021-05-02 by Nurrl