plerup/makeEspArduino #163: flash not working for ESP32

enuit, 2021-05-04:
I use your Makfile for building and flashing my ESP32. However, the flashing part doesn't work. I use it like this:
`espmake32 LIBS=../../ BOARD="esp32" flash`
and it only tells me I should use esptool different. When I look at what command Make attempts to run (`espmake32 LIBS=../../ BOARD="esp32" flash -n`), it tells me:
`python "<SOME PATH THAT LEADS TO THE ESPTOOL>/esptool/esptool.py"`

Alas, I don't know where to start searching for the problem.

plerup, 2021-05-05:
Please provide the full error message

enuit, 2021-05-05:
Well, the error message is basically what esptool returns when you call it without parameters:
  ```
Build complete. Elapsed time: 21 seconds
  
  esptool.py v3.1-dev
  usage: esptool [-h] [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3}] [--port PORT] [--baud BAUD] [--before {default_reset,no_reset,no_reset_no_sync}]
                 [--after {hard_reset,soft_reset,no_reset}] [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]] [--connect-attempts CONNECT_ATTEMPTS]
                 {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                 ...
  
  esptool.py v3.1-dev - ESP8266 ROM Bootloader Utility
  
  positional arguments:
    {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,version,get_security_info}
                          Run esptool {command} -h for additional help
      load_ram            Download an image to RAM and execute
      dump_mem            Dump arbitrary memory to disk
      read_mem            Read arbitrary memory location
      write_mem           Read-modify-write to arbitrary memory location
      write_flash         Write a binary blob to flash
      run                 Run application code in flash
      image_info          Dump headers from an application image
      make_image          Create an application image from binary files
      elf2image           Create an application image from ELF file
      read_mac            Read MAC address from OTP ROM
      chip_id             Read Chip ID from OTP ROM
      flash_id            Read SPI flash manufacturer and device ID
      read_flash_status   Read SPI flash status register
      write_flash_status  Write SPI flash status register
      read_flash          Read SPI flash content
      verify_flash        Verify a binary blob against flash
      erase_flash         Perform Chip Erase on SPI flash
      erase_region        Erase a region of the flash
      merge_bin           Merge multiple raw binary files into a single file for later flashing
      version             Print esptool version
      get_security_info   Get some security-related data
  
  optional arguments:
    -h, --help            show this help message and exit
    --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3beta3,esp32c3}
                          Target chip type
    --port PORT, -p PORT  Serial port device
    --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
    --before {default_reset,no_reset,no_reset_no_sync}
                          What to do before connecting to the chip
    --after {hard_reset,soft_reset,no_reset}, -a {hard_reset,soft_reset,no_reset}
                          What to do after esptool.py is finished
    --no-stub             Disable launching the flasher stub, only talk to ROM bootloader. Some features will not be available.
    --trace, -t           Enable trace-level output of esptool.py interactions.
    --override-vddsdio [{1.8V,1.9V,OFF}]
                          Override ESP32 VDDSDIO internal voltage regulator (use with care)
    --connect-attempts CONNECT_ATTEMPTS
                          Number of attempts to connect, negative or 0 for infinite. Default: 7.
  make: *** [<PATH TO MAKEFILE>/makeEspArduino/makeEspArduino.mk:334: flash] Fehler 1
```

plerup, 2021-05-05:
Ok, the latest commits of platform.txt in arduino-esp32 messed up things for makeEspArduino.
You have to go back to version 1.0.6 of arduino-esp32 to get it working for now
I will try to fix this soon

enuit, 2021-05-06:
That was indeed the problem! Thank you very much for the help, especially considering I didn't mention which version of the Arduino-ESP32 repository I'm using. Much appreciated.

plerup, 2021-05-14:
Fixed in latest commit

closed 2021-06-18 by plerup