plerup/makeEspArduino #169: EXCLUDE_DIRS not honored anymore

everslick, 2021-09-16:
For me, it seems, as the `EXCLUDE_DIRS` variable is not used at all, anymore.

I set:

```
EXCLUDE_DIRS      ?= $(ESP_LIBS)/DNSServer      \
                     $(ESP_LIBS)/ESP8266SdFat   \
                     $(ESP_LIBS)/LittleFS/lib
```

Still it tries to compile:

```
lfs_filebd.c
/home/clemens/Devel/ESP/Arduino-ESP8266/libraries/LittleFS/lib/littlefs/bd/lfs_filebd.c:7:27: fatal error: bd/lfs_filebd.h: No such file or directory
 #include "bd/lfs_filebd.h"
                           ^
compilation terminated.
make[1]: *** [../../makeEspArduino/makeEspArduino.mk:274: ../build/3.0.99/ahe/eng/alpha/lfs_filebd.c.o] Error 1
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/src'
make: *** [Makefile:70: src] Error 2
```

Has something changed here lately?

everslick, 2021-09-16:
Also it looks like EVERYTHING under `$(ARDUINO)/libraries` is now included for the build by default with no way to limit the build files to a certain sub-set. right?

plerup, 2021-09-17:
First question, yes the EXCLUDE_DIRS is currently not applied to the core file search. This is an error and will be fixed
Second question,   ARDUINO_LIBS can be set to anything by you but the default is to search the whole standard location. Here EXCLUDE_DIRS is also applied for filtering

everslick, 2021-09-17:
so, I should collect all ARDUINO library directories in the ARDUINO_LIBS variable, and all MY OWN libraries in the LIBS variable, right? or collect everything in LIBS and set `ARDUINO_LIBS=""`, same thing? THX

plerup, 2021-09-17:
Yes, that should do it

plerup, 2021-09-17:
Regarding first case. I understood wrong, it's not a CORE file which is your problem. These files have never been checked for exclusion

I have not been able to reproduce the error you get here. Can you perhaps provide an example?

plerup, 2021-09-17:
Ok, now I see what the problem is. The EXCLUDE_DIRS must now be a regular expression. Not clearly stated in the README, will be updated.

So change to EXCLUDE_DIRS ?= $(ESP_LIBS)/DNSServer|$(ESP_LIBS)/ESP8266SdFat|$(ESP_LIBS)/LittleFS/lib

However the LittleFS/lib directory is always excluded by default so the last directory is no longer needed

everslick, 2021-09-17:
Thank you, I thought the **regular expression** was optional. Now it works like it should! :-D

Thanks, once again, for **makeEspArduino** and for your kind support!

closed 2021-09-17 by everslick