plerup/makeEspArduino #171: ESP8266WiFi.h: No such file or directory

Animeshdhakal, 2021-09-18:
I just upgraded to new version of makeEspArduino now its giving me this error 
```
/tmp/mkESP/main_generic/main.cpp.cpp:2:10: fatal error: ESP8266WiFi.h: No such file or directory
    2 | #include <ESP8266WiFi.h>
```

plerup, 2021-09-19:
Please provide some more information about your configuration

Animeshdhakal, 2021-09-19:
> Please provide some more information about your configuration

```
import os
import requests
import sys

def help():
    c = "List of Commands: \n"
    c += "`python3 build/get.py install` for Installation\n"
    c += "`python3 build/get.py vscode` for vscode"
    c += "`python3 build/get.py makefile` for generating Makefile vscode"
    print(c)

def generateMakefile(board):
    file = open("Makefile", "w")
    content = f"ESP_ROOT = ./build/{board}/\n"
    content += "SKETCH = ./src/main.cpp\n"
    content += "MONITOR_SPEED = 115200\n"
    content += "LIBS = ./lib\n\n"
    content += "include ./build/makeEspArduino/makeEspArduino.mk"
    file.write(content)
    file.close()

def generateVSCode(board):
    if board == "8266":
        res = requests.get("https://gist.githubusercontent.com/Animeshdhakal/94f02a4dd25787e754497536a7d558ca/raw/0b8e67ad92fd1e6b11c9e5d00d756482575926b5/c_cpp_properties_esp8266.json")
    elif board == "32":
        res = requests.get("https://gist.githubusercontent.com/Animeshdhakal/94f02a4dd25787e754497536a7d558ca/raw/0b8e67ad92fd1e6b11c9e5d00d756482575926b5/c_cpp_properties_esp8266.json")
    else:
        return

    if not os.path.isdir(".vscode"):
        os.mkdir(".vscode")

    file = open(".vscode/c_cpp_properties.json", "w")
    file.write(res.text)
    file.close()
    

def install():
    board = input("Enter the Board Type (8266 for ESP8266 and 32 for ESP32): ")
    if board in ["8266", "32"]:
        cmd = "cd build \n"
        cmd += "git clone https://github.com/plerup/makeEspArduino/ \n"

        if board == "8266":
            cmd += "git clone https://github.com/esp8266/arduino esp8266 \n"

        elif board == "32":
            cmd += "git clone https://github.com/espressif/arduino-esp32 esp32 \n"

        cmd += f"cd esp{board} \n"

        version = input("Enter the Core Version (master for master version): ")

        if version != "master":
            cmd += f"git checkout tags/{version} -f \n"
        
        cmd += "git submodule update --init -f \n"
        cmd += "cd tools \n"
        cmd += "python3 get.py \n"

        os.system(cmd)
        generateMakefile("esp"+board)
        generateVSCode(board)

    else:
        print("This board is not available or Invalid")

def main():
    if len(sys.argv) >= 2:
        if sys.argv[1] == 'install':
            install()
        elif sys.argv[1] == 'vscode':
            board = input("Enter the Board Type (8266 for ESP8266 and 32 for ESP32): ")
            if board in ["8266", "32"]:
                generateVSCode(board)
            else:
                print("Invalid Board")
        else:
            help()
    else:
        help()

if __name__ == "__main__":
    main()
```

This is the python file used to install and generate makefile

plerup, 2021-09-20:
The definition of ESP_ROOT must be an absolute path. You have to resolve it either in the python script or something like this in the generated makefile

$(dir $(abspath $(firstword $(MAKEFILE_LIST))))

Animeshdhakal, 2021-09-21:
Thanks for the help

closed 2021-09-21 by Animeshdhakal

MatthiasPortzel, 2022-01-11:
It appears to me that ESP_ROOT still needs to be an absolute path. I wrote out my individual set up in #179, including `ESP_ROOT="$(realpath esp8266)"`, because it wouldn't compile without it (even regardless of the bracket issue).