plerup/makeEspArduino #177: Couldn't write a littlefs data filesystem

lugdunum-1964, 2021-12-19:
For a project I'm storing a web site on a littlefs data partition on the chip.
During make flash_fs I found two issues:

first:
esptool.py in ~/esp32/tools/esptool didn't exist, was called esptool without extension,
I created a symlink to solve.

second:
parse_arduino.pl created the the same command for FS_UPLOAD_COM as for UPLOAD_COM. 
I traced that to line 150:
    `$fs_upload_com =~ s/(.+ --flash_size detect) .+/$1 \$(SPIFFS_START) \$(FS_IMAGE)/;`
    
The flash size in platform.txt and boards.txt is always 2, 4, 8, or 16MB, so the search pattern 'detect'  always failed.  This maent running "make flash_fs" just flashed the code, and not the data.

Replacing the line by 
   `$fs_upload_com =~ s/(.+ --flash_size \d+MB) .+/$1 \$(SPIFFS_START) \$(FS_IMAGE)/;`
fixed the issue

plerup, 2021-12-20:
Thanks for reporting. I missed testing for esp32. Fix coming soon

plerup, 2021-12-20:
@lugdunum-1964
Well, this works for me. Which version of esp32 Arduino are you using and have you run get.py in the tools directory?

plerup, 2021-12-20:
@lugdunum-1964 
Regarding the first point above, did you run the get.py in the tools directory? That commands creates the esptool sub directory

lugdunum-1964, 2021-12-20:
Regarding running get.py, I did run that.
Regarding the esp32, I'm using a AZdelivery ESP32 Dev Kit V4, which according to the manual should be treated as a regular esp32 dev kit, so board=esp32, chip=esp32.

I checked both boards.txt and platfom.txt in the esp32 directory (the latest pull from github) and none of the boards have a build.flash_size=detect, it's all 2,4,8 or 16MB. So the replacement regex wouldn't match any of these.

plerup, 2021-12-21:
Fixed in latest commit now

lugdunum-1964, 2021-12-21:
Thanks, works as expexted now

closed 2021-12-21 by lugdunum-1964