plerup/makeEspArduino #178: Specifying multiple library directories and source files in LIBS seems not to work

hamishcunningham, 2022-01-10:
hi again :)

I tried to include both a directory tree and a specific file from another tree in LIBS, but the latter doesn't get included in the build for some reason. E.g.:

```bash
make -f .../makeEspArduino/makeEspArduino.mk ESP_ROOT=.../Arduino/hardware/espressif/esp32     BOARD=featheresp32 CHIP=esp32 LIBS="...unphone/lib .../unphone/lib/unPhone_Library/src/IOExpander.cpp" SKETCH=sketch/sketch.ino  run
```

This gets me undefined symbol errors and IOExpander.cpp isn't present in /tmp/mkESP

plerup, 2022-01-10:
Why are there three dots in the path?

hamishcunningham, 2022-01-10:
I just abbreviated them. Everything is in $HOME

plerup, 2022-01-10:
Which sketch is this in the repo?

hamishcunningham, 2022-01-10:
SdInfo; here's the full command:

```bash
make -f /home/hamish/the-internet-of-things/support/tooling/makeEspArduino/makeEspArduino.mk \
  ESP_ROOT=/home/hamish/the-internet-of-things/support/tooling/Arduino/hardware/espressif/esp32 \
  BOARD=featheresp32 CHIP=esp32 \
  LIBS="/home/hamish/unphone/lib /home/hamish/unphone/src/IOExpander.cpp" \
  SKETCH=/home/hamish/unphone/lib/SdFat/examples/SdInfo/SdInfo.ino  run
```

plerup, 2022-01-10:
Strange, I pasted this very command and it built without any problems

...
esp32-hal-bt.c
HardwareSerial.cpp
1_IOExpander.cpp
1_WebServer.cpp
2_IOExpander.cpp
2_WebServer.cpp
SdInfo.ino.cpp
Creating core archive
Creating object archive
Linking /tmp/mkESP/SdInfo_featheresp32/SdInfo.bin
  Versions: 0cbe2d8-dirty, 2.0.2
esptool.py v3.1
Merged 2 ELF sections

Memory summary
  RAM:    17172 bytes
  Flash: 244925 bytes

hamishcunningham, 2022-01-10:
Strange! Did you do a clean first? Apart from that I'm on 21.04, maybe it is some subtle difference in the OS. (I also see with a fresh install that erase_flash doesn't work until I link esptool.py to esptool; perhaps we've got different versions of the core?)

plerup, 2022-01-10:
Well, this is a virtual machine in which I installed from scratch according to your instructions in the previous case.
What is it actually that you want to do here? Is it to replace the IOExpander file or to use both of them?

hamishcunningham, 2022-01-10:
We develop a device that requires a patched version of digitalWrite (so we can toggle chip select via an io expander), so I need to trigger compilation and linking of the various patched libraries and the expander code and the sketch.

hamishcunningham, 2022-01-10:
PS if I copy in the expander code to the sketch directory the Arduino IDE will compile and link. I can't figure out how to achieve the same with makeEspArduino

plerup, 2022-01-11:
Sorry but I still don't get it. By specifying:

LIBS="/home/hamish/unphone/lib /home/hamish/unphone/src/IOExpander.cpp"

you will get two files in the wildcard search:
unphone/src/IOExpander.cpp
unphone/lib/unPhone_Library/IOExpander.cpp

As far as I can see they are both identical.

However when it comes to patching, i.e. replacing a possible existing file, you need to use the variable EXCLUDE_DIRS, see more in the readme

plerup, 2022-01-11:
Is this a new case?
I'm sorry but I can't test more than the example you provided

LIBS can contain either directories which means an implicit wildcard for all sub directories, explicit wildcards or specific full file paths. All source files found when evaluating this will be added to the build. When I build your case I get the following (via new logging I just added)

- Finding all involved files for the build ...
* Duplicate
    /home/peter/unphone/lib/ESPAsyncWebServer/src/WebServer.cpp
  copied to
    $(BUILD_DIR)/1_WebServer.cpp
* Duplicate
    /home/peter/the-internet-of-things/support/tooling/Arduino/hardware/espressif/esp32/libraries/WebServer/src/WebServer.cpp
  copied to
    $(BUILD_DIR)/2_WebServer.cpp
* Duplicate
    /home/peter/unphone/src/IOExpander.cpp
  copied to
    $(BUILD_DIR)/1_IOExpander.cpp
* Duplicate
    /home/peter/unphone/lib/unPhone_Library/IOExpander.cpp
  copied to
    $(BUILD_DIR)/2_IOExpander.cpp

I can assure you that individual files get added when present i LIBS, that is something that I use extensively myself

hamishcunningham, 2022-01-11:
I span up a 20:04 docker image and ran the command as root and it behaved differently to my local system. Will try to duplicate again tomorrow!  Tnx, H

closed 2022-06-08 by plerup