plerup/makeEspArduino #180: LittleFS compile errors on esp8266

KaDw, 2022-01-12:
Great project!

I have a sketch that is using LittleFS and esp8266 arduino core 3.0.2. In sketch I include <FS.h> and <LittleFS.h>. To compile the project I use following command: `make -f /home/karol/makeEspArduino/makeEspArduino.mk ESP_ROOT=/home/karol/esp8266 flash` inside the folder that has .ino file.

I get following output:
```
~/code/esp8266-test/esp8266-test$ make -f /home/karol/makeEspArduino/makeEspArduino.mk ESP_ROOT=/home/karol/esp8266 flash
2_lfs.c
ccache "/home/karol/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-I/home/karol/esp8266/tools/sdk/include" "-I/home/karol/esp8266/tools/sdk/lwip2/include" "-I/home/karol/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/home/karol/code/esp8266-test/esp8266-test/build/core"  -c -w -Werror=return-type -std=gnu17  -Os -g -free -fipa-pta -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_GENERIC\"  -DFLASHMODE_DOUT  -DESP8266  -I/home/karol/esp8266/cores/esp8266 -I/home/karol/esp8266/variants/generic -I/home/karol/code/esp8266-test/esp8266-test/build -I/home/karol/code/esp8266-test/esp8266-test -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/extras/tests/Helpers -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/extras/tests/Helpers/api -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/StringStorage -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Polyfills -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Polyfills/type_traits -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Polyfills/mpl -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Memory -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Numbers -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Deserialization -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Deserialization/Readers -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Collection -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Variant -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Object -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Json -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Document -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/MsgPack -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Array -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Strings -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Strings/Adapters -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Misc -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Serialization -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Serialization/Writers -I/home/karol/code/esp8266-test/esp8266-test/lib/WiFiManager -I/home/karol/code/esp8266-test/esp8266-test/lib/OneButton/src -I/home/karol/esp8266/libraries/ESP8266HTTPClient/src -I/home/karol/esp8266/libraries/ESP8266WiFi/src -I/home/karol/esp8266/libraries/ESP8266WiFi/src/include -I/home/karol/esp8266/libraries/ESP8266HTTPUpdateServer/src -I/home/karol/esp8266/libraries/ESP8266WebServer/src -I/home/karol/esp8266/libraries/ESP8266WebServer/src/detail -I/home/karol/esp8266/libraries/Wire -I/home/karol/esp8266/libraries/DNSServer/src -I/home/karol/esp8266/libraries/ESP8266mDNS/src -I/home/karol/esp8266/libraries/GDBStub/src -I/home/karol/esp8266/libraries/GDBStub/src/internal -I/home/karol/esp8266/libraries/GDBStub/src/xtensa -I/home/karol/esp8266/libraries/GDBStub/src/xtensa/config -I/home/karol/esp8266/libraries/Ticker/src -I/home/karol/esp8266/libraries/LittleFS/src    /home/karol/code/esp8266-test/esp8266-test/build/2_lfs.c -o /home/karol/code/esp8266-test/esp8266-test/build/2_lfs.c.o
2_lfs_util.c
ccache "/home/karol/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-I/home/karol/esp8266/tools/sdk/include" "-I/home/karol/esp8266/tools/sdk/lwip2/include" "-I/home/karol/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/home/karol/code/esp8266-test/esp8266-test/build/core"  -c -w -Werror=return-type -std=gnu17  -Os -g -free -fipa-pta -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10605 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_GENERIC\"  -DFLASHMODE_DOUT  -DESP8266  -I/home/karol/esp8266/cores/esp8266 -I/home/karol/esp8266/variants/generic -I/home/karol/code/esp8266-test/esp8266-test/build -I/home/karol/code/esp8266-test/esp8266-test -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/extras/tests/Helpers -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/extras/tests/Helpers/api -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/StringStorage -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Polyfills -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Polyfills/type_traits -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Polyfills/mpl -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Memory -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Numbers -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Deserialization -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Deserialization/Readers -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Collection -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Variant -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Object -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Json -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Document -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/MsgPack -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Array -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Strings -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Strings/Adapters -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Misc -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Serialization -I/home/karol/code/esp8266-test/esp8266-test/lib/ArduinoJson/src/ArduinoJson/Serialization/Writers -I/home/karol/code/esp8266-test/esp8266-test/lib/WiFiManager -I/home/karol/code/esp8266-test/esp8266-test/lib/OneButton/src -I/home/karol/esp8266/libraries/ESP8266HTTPClient/src -I/home/karol/esp8266/libraries/ESP8266WiFi/src -I/home/karol/esp8266/libraries/ESP8266WiFi/src/include -I/home/karol/esp8266/libraries/ESP8266HTTPUpdateServer/src -I/home/karol/esp8266/libraries/ESP8266WebServer/src -I/home/karol/esp8266/libraries/ESP8266WebServer/src/detail -I/home/karol/esp8266/libraries/Wire -I/home/karol/esp8266/libraries/DNSServer/src -I/home/karol/esp8266/libraries/ESP8266mDNS/src -I/h [... cut ...]
THIS_DIR := $(realpath $(dir $(realpath $(lastword $(MAKEFILE_LIST)))))
ROOT := $(THIS_DIR)/..
LIBS =  \
    $(ROOT)/lib/ArduinoJson \
    $(ROOT)/lib/WiFiManager \
    $(ROOT)/lib/OneButton

FLASH_DEF = 4M1M
FS_TYPE=LittleFS
BUILD_DIR ?= $(shell pwd)/build

VERBOSE = 1
USE_CCACHE = 1

COM = ttyUSB0
UPLOAD_SPEED = 921600
```

I'm not sure why there is 2_lfs.c and 2_lfs_util.c in my build folder. It seems that 2_lfs_util.c is a default config file, not sure what 2_lfs.c does.

Any ideas what I did wrong?

plerup, 2022-01-13:
Fixed in latest commit

KaDw, 2022-01-13:
Thank you, latest master works great! 

I also noticed small bug in my config.mk, instead of `FS_TYPE=LittleFS` I should use `FS_TYPE=littlefs` because otherwise I get following error:
```
Generating file system image: /home/karol/code/esp8266-test/esp8266-test/build/FS.bin
"" -b 8192 -s 0xFA000 -c /home/karol/code/esp8266-test/esp8266-test/data /home/karol/code/esp8266-test/esp8266-test/build/FS.bin
/bin/sh: 1: : Permission denied
make: *** [/home/karol/makeEspArduino/makeEspArduino.mk:355: /home/karol/code/esp8266-test/esp8266-test/build/FS.bin] Error 127
The terminal process "/bin/bash '-c', 'make -f /home/karol/makeEspArduino/makeEspArduino.mk ESP_ROOT=/home/karol/esp8266 flash_fs '" failed to launch (exit code: 2).
```

plerup, 2022-06-08:
Fixed now

closed 2022-06-08 by plerup