plerup/makeEspArduino #183: ESP32 python: can't open file '/esptool.py': [Errno 2] No such file or directory

Blackqpid, 2022-05-14:
Running the latest commit (5416111), I'm getting the following:

```
$ make -f makeEspArduino.mk DEMO=1 CHIP=esp32             
* Build state has changed, doing a full rebuild *
- Finding all involved files for the build ...
- Parsing Arduino configuration files ...
ETH.cpp
WiFi.cpp
...
stdlib_noniso.c
WiFiScan.ino.cpp
Creating core archive
Creating object archive
Linking /tmp/mkESP/WiFiScan_esp32/WiFiScan.bin
  Versions: Unknown, 1.0.6
python: can't open file '/esptool.py': [Errno 2] No such file or directory
make: *** [makeEspArduino.mk:315: /tmp/mkESP/WiFiScan_esp32/WiFiScan.bin] Error 2
```

esptool.py is in:
 ```
$ find .arduino15 -name "*esptool*" 
.arduino15/staging/packages/esptool-3.0.0.2-linux.tar.gz
.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/esptool.py
.arduino15/packages/esp32/tools/esptool_py
.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py
.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool
.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/flasher_stub/esptool_test_stub.py
.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/test/test_esptool.py
.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/__pycache__/esptool.cpython-39.pyc
.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/esptool.py
```

I'm using the Arduino IDE (1.8.19) and esp32 board from Espressif 1.0.6 installed. I'm running this on Linux. The ESP8266 demo builds without issue, just the ESP32 has an issue.

Any help would be appreciated. I've been using the makefile for years for ESP8266 and it's a great help. I just need the ESP32 to work now.

hamishcunningham, 2022-05-14:
I think I've seen this too occasionally; a workaround is to
cd .arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/esptool/
ln -s esptool.py esptool
(or something similar, depending on your platform)
HTH
H

Blackqpid, 2022-05-14:
I've just tried that, and get the same error.
I think it's trying to run esptool.py located in the root (/) directory, which doesn't exist, so I guess the makefile is missing off the path to esptool when trying to run it.

plerup, 2022-05-15:
The handling of esptool is a bit complex as it has changed a lot during the development of both Arduino ESP8266 and Arduino ESP32.
However it works for me on both cases. You can always force the path to esptool by supplying the variable ESPTOOL on the make command line, e.g.
ESPTOOL=$HOME/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py

Blackqpid, 2022-05-15:
I've tried setting it on the command line as suggested, and I still get the same error. Using the trace option of make, shows that for the other commands make is running, it's using the full path, but for esptool.py it's ignoring the path and thinks esptool.py is in the root directory.

```
$ make -f makeEspArduino.mk DEMO=1 CHIP=esp32 ESPTOOL=$HOME/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py --trace
...
python "/home/blackqpid/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/gen_esp32part.py" -q /home/blackqpid/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/default.csv "/tmp/mkESP/WiFiScan_esp32/WiFiScan.partitions.bin"
python "/esptool.py" --chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size "4MB" -o "/tmp/mkESP/WiFiScan_esp32/WiFiScan.bin" "/tmp/mkESP/WiFiScan_esp32/WiFiScan.elf"
python: can't open file '/esptool.py': [Errno 2] No such file or directory
make: *** [makeEspArduino.mk:315: /tmp/mkESP/WiFiScan_esp32/WiFiScan.bin] Error 2
```

plerup, 2022-05-16:
The handling of the tools directories in Arduino ESP32 is a real mess when using native an Arduino installation.
I'm working on a solution.
However if you use a git version of Arduino ESP32 as source it will work with the current version of makeEspArduino

plerup, 2022-06-08:
Fixed in latest commit

closed 2022-06-08 by plerup