plerup/makeEspArduino #187: Arch WSL g++

waynewec, 2022-10-13:
Running a makefile including a config (sets OS_NAME and ARDUINO_ROOT) and makeEspArduino.mk through VsCode with the WSL plugin to an Arch Linux install. Arduino installed on Windows host and works correctly.

Result of `make DEMO=1`
```
DESKTOP-PVKV5V9% make DEMO=1
abi.cpp
aes_unwrap.cpp
base64.cpp
/bin/sh: line 1: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++: No such file or directory
/bin/sh: line 1: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++: No such file or directory
cbuf.cpp
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:278: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/abi.cpp.o] Error 127
make: *** Waiting for unfinished jobs....
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:278: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/aes_unwrap.cpp.o] Error 127
/bin/sh: line 1: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++: No such file or directory
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:279: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/base64.cpp.o] Error 127
/bin/sh: line 1: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++: No such file or directory
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:279: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/cbuf.cpp.o] Error 127
```
After making a copy of the `xtensa-lx106-elf-g++` without the `.exe` file extension
```
DESKTOP-PVKV5V9% make DEMO=1
* Build state has changed, doing a full rebuild *
- Finding all involved files for the build ...
* Warning: Ignoring non existing file specification E:wayneDocumentsArduino/libraries
- Parsing Arduino configuration files ...
BearSSLHelpers.cpp
CertStoreBearSSL.cpp
ESP8266WiFi.cpp
ESP8266WiFiAP.cpp
xtensa-lx106-elf-g++: error: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/BearSSLHelpers.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
ESP8266WiFiGeneric.cpp
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:278: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/BearSSLHelpers.cpp.o] Error 1
make: *** Waiting for unfinished jobs....
xtensa-lx106-elf-g++: error: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/CertStoreBearSSL.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:279: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/CertStoreBearSSL.cpp.o] Error 1
xtensa-lx106-elf-g++: error: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp: No such file or directory
                                               xtensa-lx106-elf-g++: fatal error: no input files
                                                                                                compilation terminated.
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:279: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/ESP8266WiFi.cpp.o] Error 1
                                                              xtensa-lx106-elf-g++: error: /mnt/c/Users/wayne/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp: No such file or directory
                                                                                                               xtensa-lx106-elf-g++: fatal error: no input files
                           compilation terminated.
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:279: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/ESP8266WiFiAP.cpp.o] Error 1
                                                                xtensa-lx106-elf-g++: error: /mnt/c/Users/wayne/AppData/Local/Arduino
15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
make: *** [/mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/makeEspArduino/makeEspArduino.mk:279: /mnt/e/Projects/ArduinoCode/Arduino_VsCode_WSL_Framework/build/ESP8266WiFiGeneric.cpp.o] Error 1
```

plerup, 2023-05-15:
Late response, sorry. I assume you still have this problem?

plerup, 2023-06-07:
Closing due to lack of response

closed 2023-06-07 by plerup