plerup/makeEspArduino #188: Calling 'make clean' twice will trigger config parser.

everslick, 2022-11-07:
This is really only a minor issue, but maybe it can be fixed with relative ease.

```
clemens@alien:~/Devel/ESP/emonio-fw$ make clean 
make[1]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/html'
rm -rf ../build/3.0.99/xuan/eng/alpha/html
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/html'
make[1]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/src'
Removing all build files
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/src'
clemens@alien:~/Devel/ESP/emonio-fw$ make clean 
make[1]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/html'
rm -rf ../build/3.0.99/xuan/eng/alpha/html
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/html'
make[1]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/src'
- Parsing Arduino configuration files ...
- Finding all involved files for the build ...
* Duplicate:
    /home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/nimble/nimble/drivers/nrf52/src/ble_phy.c
  copied to:
    $(BUILD_DIR)/1_ble_phy.c
* Duplicate:
    /home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/nimble/nimble/drivers/nrf51/src/ble_phy.c
  copied to:
    $(BUILD_DIR)/2_ble_phy.c
* Duplicate:
    /home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/nimble/nimble/drivers/nrf51/src/ble_hw.c
  copied to:
    $(BUILD_DIR)/1_ble_hw.c
* Duplicate:
    /home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/nimble/nimble/drivers/nrf52/src/ble_hw.c
  copied to:
    $(BUILD_DIR)/2_ble_hw.c
- Parsing Arduino configuration files ...
Removing all build files
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/src'
```

I think this is because the $(BUILD_DIR) gets deleted on the first **clean** and that triggers the rules that have it as prerequisite.

everslick, 2022-11-07:
Also a minor thing is, that the file search will find duplicated files even in directories that are in **$(EXCLUDE_DIRS)**. Besides I *think* EXCLUDE_DIRS is not honored consistently (i.e. I see dirs in `list_lib` that should be excluded).

plerup, 2022-11-07:
Can you please provide example on your setting of EXCLUDE_DIRS and a false positive from that?

everslick, 2022-11-07:
```
EXCLUDE_DIRS      :=  $(ESP_LIBS)/DNSServer                                     
EXCLUDE_DIRS      += |$(ESP_LIBS)/ESP8266SdFat                                  
EXCLUDE_DIRS      += |$(ESP_LIBS)/Ethernet                                      
EXCLUDE_DIRS      += |$(ESP_LIBS)/AzureIoT                                      
EXCLUDE_DIRS      += |$(ESP_LIBS)/ArduinoOTA                                    
EXCLUDE_DIRS      += |$(ESP_LIBS)/BluetoothSerial                               
```

```
clemens@alien:~/Devel/ESP/emonio-fw$ make list_lib 
make[1]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/html'
make[1]: Nothing to be done for 'list_lib'.
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio-fw/html'
make[1]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/src'
===== Include directories =====
/home/clemens/Devel/ESP/emonio-fw/src
/home/clemens/Devel/ESP/emonio-fw/libs/WebServer
/home/clemens/Devel/ESP/emonio-fw/libs/WebSockets
/home/clemens/Devel/ESP/emonio-fw/libs/WebUpdate
/home/clemens/Devel/ESP/emonio-fw/libs/CaptivePortal
/home/clemens/Devel/ESP/emonio-fw/libs/Modbus
/home/clemens/Devel/ESP/emonio-fw/libs/OLEDDisplay
/home/clemens/Devel/ESP/emonio-fw/libs/MHZ19
/home/clemens/Devel/ESP/emonio-fw/libs/Haru
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/Wire/src
/home/clemens/Devel/ESP/emonio-fw/libs/Hash/sha1
/home/clemens/Devel/ESP/emonio-fw/libs/Hash
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/Update/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/HTTPClient/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFiClientSecure/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/ESPmDNS/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/FS/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/FFat/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SD/src
/home/clemens/Devel/ESP/emonio-fw/libs/Servo
/home/clemens/Devel/ESP/emonio-fw/libs/NimBLE
/home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/nimble/ext/tinycrypt/src
**(cutting lots of NimBLE DIRS here)**
/home/clemens/Devel/ESP/emonio-fw/libs/BLEDeviceID
/home/clemens/Devel/ESP/emonio-fw/libs/BME680
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SPI/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/ArduinoOTA/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/BluetoothSerial/src
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SPIFFS/src
===== Source files =====
../build/3.0.99/xuan/eng/alpha/1_ble_hw.c
../build/3.0.99/xuan/eng/alpha/1_ble_phy.c
../build/3.0.99/xuan/eng/alpha/2_ble_hw.c
../build/3.0.99/xuan/eng/alpha/2_ble_phy.c
/home/clemens/Devel/ESP/emonio-fw/libs/BLEDeviceID/BLEDeviceID.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/BME680/bme680.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/BME680/bme680_drv.c
/home/clemens/Devel/ESP/emonio-fw/libs/CaptivePortal/DNSServer.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/Modbus/ModbusProto.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/Modbus/ModbusRTU.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/Modbus/ModbusTCP.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/NimBLE2904.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/NimBLE/NimBLEAddress.cpp
**(cutting lots of NimBLE FILES here)**
/home/clemens/Devel/ESP/emonio-fw/libs/OLEDDisplay/OLEDDisplay.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/OLEDDisplay/OLEDDisplayUi.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/Servo/ESP32PWM.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/Servo/ESP32Servo.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/WebServer/ESPWebServer.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/WebServer/Parsing.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/WebServer/mimetable.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/WebSockets/WebSockets.cpp
/home/clemens/Devel/ESP/emonio-fw/libs/WebSockets/WebSocketsClient.cpp
[... 10 lines cut ...]
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/FS/src/vfs_api.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/HTTPClient/src/HTTPClient.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SD/src/SD.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SD/src/sd_diskio.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SD/src/sd_diskio_crc.c
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SPI/src/SPI.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/SPIFFS/src/SPIFFS.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/Update/src/HttpsOTAUpdate.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/Update/src/Updater.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFi.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFiAP.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFiClient.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFiGeneric.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFiMulti.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFiSTA.cpp
[... cut ...]
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFi/src/WiFiUdp.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFiClientSecure/src/esp_crt_bundle.c
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/WiFiClientSecure/src/ssl_client.cpp
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-2.0.0/libraries/Wire/src/Wire.cpp
```

BluetoothSerial and ArduinoOTA should not be there, no?

plerup, 2022-11-07:
The problem here is your declaration of EXCLUDE_LIBS. This must be a strict regexp, you have both trailing spaces when setting the variable and += will always add a space as well. This means that there will be no match.
Also depending on where you put it in the makefile chain, ESP_LIBS may not have been defined.

everslick, 2022-11-07:
ah, IC. THX. that's a good hint. ESP_LIBS is not to blame, but the spaces are a problem of course. will dig into it.

everslick, 2022-11-07:
ok, this is tricky when one, like me, uses compile time feature flags a lot. this is an excerpt from my solution (maybe it helps someone):

```
# collect libraries to exclude from build                                       

EXCL = $(ESP_LIBS)/DNSServer            \
       $(ESP_LIBS)/ESP8266SdFat         \
       $(ESP_LIBS)/Ethernet             \
       $(ESP_LIBS)/AzureIoT

ifeq ($(HAVE_ARDUINO_OTA),0)
EXCL += $(ESP_LIBS)/ArduinoOTA
endif

ifeq ($(HAVE_SPP),0)
EXCL += $(ESP_LIBS)/BluetoothSerial
endif

ifeq ($(HAVE_LITTLEFS),0)
EXCL += $(ESP_LIBS)/LittleFS
endif

null :=
space := ${null} ${null}
${space} := ${space}

EXCLUDE_DIRS = $(subst ${ },|,$(strip $(EXCL)))
```

works as advertised. THANK YOU!

plerup, 2022-11-08:
Well, actually I guess the removal of spaces could be done automatically in makeEspArduino as you would normally (and shouldn't) use spaces in path names.
Anyhow, glad you got it working

closed 2023-06-06 by plerup