plerup/makeEspArduino #189: build_opt.h: No such file or directory

lugdunum-1964, 2022-11-14:
After a "make clean" or during the first make I get an error:
build_opt.h: No such file or directory

The culprit appears to be the default multi-threaded make. Setting BUILD_THREADS=1 or running "make -j 1" fixes the issue.

Maybe there is a dependency missing somewhere?

--
Regards Lyon Lemmens

everslick, 2022-11-14:
I have the same, but I worked around it by sticking

`BUILD_OPT_H := $(shell touch $(BUILD_DIR)/build_opt.h)`

in my Makefile.

lugdunum-1964, 2022-11-14:
Thanks!! That almost works. BUILD_DIR may not exist, so I changed it to:

`BUILD_OPT_H := $(shell [ -d $(BUILD_DIR) ] || mkdir -p $(BUILD_DIR); touch $(BUILD_DIR)/build_opt.h              )`

plerup, 2022-11-14:
Ok, that was an ugly prebuild thingy in platform.txt. Will fix

lugdunum-1964, 2022-11-17:
Cheers!

plerup, 2023-05-14:
Finally got around to fix this. Please check latest version.