plerup/makeEspArduino #190: Duplicate SKRIPT when including on working directory

maximweb, 2022-11-27:
It may be more of a cosmetic thing, but I noticed that my main script gets processed twice (Duplicate warning) as soon as i include a `config.h` file from base folder.

### My minimal test example

`Makefile`:
```
SKETCH=app.cpp
include ./libs/makeEspArduino/makeEspArduino.mk
```

`app.cpp` in basefolder:
```
#include "config.h"     // if I remove this, the duplicate warning disappears
#include <Arduino.h>

void setup() {}

void loop() {}
```

`config.h` in basefolder:
```
#define VERSION_STRING "2034"
```

#### Output (excerpt)
```
* Duplicate:
    <redacted>/example/app.cpp
  copied to:
    $(BUILD_DIR)/1_app.cpp
* Duplicate:
    app.cpp
  copied to:
    $(BUILD_DIR)/2_app.cpp
```

After moving config.h to a subfolder, duplication warning disappears

#### Question
While it does not affect my build, I was still wondering if this is normal/expected behavior, and if it's possible to avoid duplacte processing of my main script while keeping the file structure identical?

plerup, 2023-05-15:
Sorry for very late response, I haven't been attending this repo for a while. It is a bug which appears when specifying the SKETCH without path. Fix is coming

maximweb, 2023-05-27:
Thank you for addressing this issue.

I saw that you made some changes and updated to version 6.8.1. Unfortunately, the issue remains unchanged.
```
- Finding all involved files for the build ...
* Duplicate:
    <full path redacted>/app.cpp
  copied to:
    $(BUILD_DIR)/1_app.cpp
* Duplicate:
    app.cpp
  copied to:
    $(BUILD_DIR)/2_app.cpp
```

plerup, 2023-05-29:
This is a bug but you can get around it by supplying the full path to the sketch in SKETCH=

maximweb, 2023-05-29:
The following works for me without the necessity to provide the full path in plain text:
`SKETCH = ${CURDIR}/app.cpp`

plerup, 2023-06-07:
Handled now in latest version

closed 2023-07-02 by plerup