plerup/makeEspArduino #192: Problems with include path

tpso1965, 2022-12-29:
After upgrading to current version compiling fails because include files in ./libraries/xxx/src is not in include path any longer.

I have a project with a local version of a mqtt-library pleaced in ./libraries.  Ths sources in ./src contains include written as
#include <PubSubClient.h>.

This worked fine with my previous version of makeEspArduino - but after update compilation fails with
PubSubClient.h not found.

plerup, 2023-05-16:
Please provide some more details about your setup

tpso1965, 2023-05-17:
All my projects have a standard set with the following directory layout:

projectX/src    (directory with the project source-files)
projectX/libraries   (extra libraries local for THIS project)
projectX/makefile

I have attached 3 makefiles:  makefile.org  that was the simple one that worked before upgrade, and a
hacked one, where I have to specify each library, to get it working.

[makefile_org.txt](https://github.com/plerup/makeEspArduino/files/11494575/makefile_org.txt)

[makefile_hacked.txt](https://github.com/plerup/makeEspArduino/files/11494576/makefile_hacked.txt)

plerup, 2023-05-29:
Try changing the CUSTOM_LIBS definition to the full path or at least ./libraries

tpso1965, 2023-05-30:
Tried to set CUSTOM_LIBS to both ./libraries and full path   - no difference.
If I dont set INCLUDE_DIRS for each library I get compile error like:

fatal error: Adafruit_Sensor.h: No such file or directory
 #include <Adafruit_Sensor.h>

If I set alle the INCLUDE_DIRS it compiles fine.

The error occur while the makefile compiles the **library** - I dont know if that is worth noting - Is compiling the libraries handled diffrently from compiling the project source ?

plerup, 2023-06-06:
This works for me at least:

    CUSTOM_LIBS = $(realpath libraries)

tpso1965, 2023-06-06:
The "realpath" did it - it worked.

Thank you!

plerup, 2023-06-07:
Used automatically in latest version

closed 2023-07-02 by plerup