plerup/makeEspArduino #194: Compilation error on core's version 3.1.0

dawidchyrzynski, 2023-02-11:
Hi,

After upgrading the ESP8266 core to version 3.1.0, my build stopped working. Upon investigation, I discovered that the ARDUINO_BOARD_ID is empty, which is causing the following error:

```
/.../libraries/ESP8266mDNS/src/LEAmDNS.cpp:1275:71: error: expected primary-expression before ')' token
 1275 |                 || (!addServiceTxt(hService, "board", ARDUINO_BOARD_ID))
```

The cpp files are compiled with the following defines:

```
-DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD=\"ESP8266_GENERIC\" -DARDUINO_BOARD_ID=  -DFLASHMODE_DOUT
```

It may be linked to this MR: https://github.com/esp8266/Arduino/pull/8634

plerup, 2023-05-16:
Sorry for late response.
I can't reproduce this error. Do you still have it? If so then please provide more details about your setup

closed 2023-06-06 by plerup

NathanV75, 2023-11-17:
@plerup I  am also experiencing this issue. I am quite sure it is related to the MR @dawidchyrzynski linked.
The generated `arduino.mk` file is has an empty value for `ARDUINO_BOARD_ID`.

This seems to only be used for the mDNS library, so you will need to include it to reproduce this.
I was able to preproduce it with the following simple sketch and using makeEspArduino v6.9.3 and esp8266 arduino v3.1.2
```
#include <ESP8266mDNS.h>

void setup() {}

void loop() {}
```

Error output:
```
...
/home/<user>/esp8266/libraries/ESP8266mDNS/src/LEAmDNS.cpp: In member function 'const void* esp8266::MDNSImplementation::MDNSResponder::enableArduino(uint16_t, bool)':
/home/<user>/esp8266/libraries/ESP8266mDNS/src/LEAmDNS.cpp:1275:71: error: expected primary-expression before ')' token
 1275 |                 || (!addServiceTxt(hService, "board", ARDUINO_BOARD_ID))
      |            
...
```

Hopefully this is enough for you to reproduce this.

plerup, 2023-11-17:
Strange, this should have taken care of that I would assume: https://github.com/esp8266/Arduino/blob/5e0ed3a38bbe9a1e203e39eb4734dc236c41e57e/libraries/ESP8266mDNS/src/LEAmDNS.cpp#L35

plerup, 2023-11-17:
OK, see it now. Thanks. Fix is coming.

plerup, 2023-11-20:
Fixed in latest commit

NathanV75, 2023-11-21:
Awesome thanks!