plerup/makeEspArduino #196: .git folder appears to be included during make

maximweb, 2023-05-27:
Only recently, after I started to add code and several libraries as `git submodule` in my project, it appears as if make tries to compile libraries from the `.git` folder in my project, resulting in multiple errors.

Adding `.git` to `EXCLUDE_DIRS` removes this issue.

I am not very familiar with how make works, but in a different project where I use a similar structure and [Arduino-Makefile](https://github.com/sudar/Arduino-Makefile), no such error occurs.

Maybe all hidden folders with leading `.` should be ignored?

plerup, 2023-05-29:
What kind of files became selected from that directory structure?

maximweb, 2023-05-29:
I use the [pubsubclient](https://github.com/knolleary/pubsubclient) library as submodule and checked out the latest version (v2.8). This is in a subfolder of my project ./libs/pubsubclient

My error message looks like this:
```
In file included from <redacted>/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp:26:
<redacted>/Code/.git/modules/libs/pubsubclient/refs/tags/list:1:1: error: expected unqualified-id before numeric constant
    1 | 2d228f2f862a95846c65a8518c79f48dfc8f188c
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

plerup, 2023-05-29:
This is strange, there should be no files getting selected in the .git directory as the search is only done for files with source file extensions. Could you please check what is in the .git/modules/libs/pubsubclient/refs/tags directory?

Excluding hidden files is of course something that makes sense, not so easy to do in an os independent way though.

maximweb, 2023-05-29:
I found a single file called `list` containing `2d228f2f862a95846c65a8518c79f48dfc8f188c`, which is basically the commit hash currently checked out in the submodule.

hamishcunningham, 2023-05-29:
Maybe the regex for the source file pattern is .c instead of \.c?

maximweb, 2023-05-29:
> I found a single file called `list` containing `2d228f2f862a95846c65a8518c79f48dfc8f188c`, which is basically the commit hash currently checked out in the submodule.

I did some more testing and found that said file was created accidentally by me when I tried to find a specific tag of the submodule. Inside the submodule's folder, I accidentally ran `git tag list`, which basically created a new tag, which then results in the file `.git/modules/libs/pubsubclient/refs/tags/list` containing the commit hash. Removing the tag removed the problem.

Yet, I am still puzzled why it is included in the first place.

plerup, 2023-05-30:
Ok, understand now. In ESP8266WiFiGeneric.cpp:26 there is:

    #include <list>

and by default all sub directories to the main sketch directory are included by default.

plerup, 2023-06-06:
Next commit will add /.git/ as default to the exclusion list

plerup, 2023-06-07:
Done now in latest version

closed 2023-07-02 by plerup