plerup/makeEspArduino #200: Add support for Arduino Core 3.0.0

everslick, 2023-10-27:
It's this time of year when major changes in the Arduino Core break makeEspArduino. But maybe this needs only a small fix. I *believe* the build breaks because the IDF headers are now in `tools/esp32-arduino-libs/esp32/` instead of `tools/sdk/esp32/`. But maybe the linking stage needs a fix too. I didn't come that far. ;-)

everslick, 2023-10-27:
changing `SDK_ROOT` to `SDK_ROOT = $(ESP_ROOT)/tools/esp32-arduino-libs` gets me further:

```
Arduino-ESP32-3.0.0/cores/esp32/chip-debug-report.cpp:212:67: error: expected primary-expression before ')' token
  212 |   chip_report_printf("  Compile Host OS   : %s\n", ARDUINO_HOST_OS);

Arduino-ESP32-3.0.0/cores/esp32/chip-debug-report.cpp:224:64: error: expected primary-expression before ')' token
  224 |   chip_report_printf("  Arduino FQBN      : %s\n", ARDUINO_FQBN);
```

I guess it's missing defines for **ARDUINO_HOST_OS** and **ARDUINO_FQBN**.

everslick, 2023-10-27:
Adding:

```
DEFINES += -DARDUINO_HOST_OS=\"$(OS)\" -DARDUINO_FQBN=\"generic\"
```

helps, but should probably done in `makeEspArduino.mk`.

plerup, 2023-10-30:
Thanks for reporting. Fix coming tonight.

plerup, 2023-11-04:
Fixed in latest commit

closed 2023-11-05 by everslick