plerup/makeEspArduino #207: FLASH_DEF ignored on esp32 ?

everslick, 2025-02-08:
I believe that on ESP32 the `FLASH_DEF=16M` is ignored and the binary is ALWAYS linked for 4MB flash size. For older `esptool.py` versions (like v3.0-dev) this did not matter and the binary could be flashed even on devices with more then 4MB. But not so with more recent versions of `esptool.py` and - by extension - with `esptool.js`. for example one will get an error like:

```
E (14) boot: load partition table error!
E (12) flash_parts: partition 3 invalid - offset 0x400000 size 0x3f0000 exceeds flash chip size 0x400000
E (12) boot: Failed to verify partition table
```

Unless I'm holding it wrong!

everslick, 2025-02-08:
More info ...

The generated `arduino.mk` file contains the line:

```
OBJCOPY="$(dir $(ESPTOOL_FILE))/esptool" --chip esp32 elf2image --flash_mode "$(FLASH_MODE)" --flash_freq "$(FLASH_SPEED)" --flash_size "4MB" --elf-sha256-offset 0xb0 -o "$(BUILD_DIR)/$(MAIN_NAME).bin" "$(BUILD_DIR)/$(MAIN_NAME).elf"
```

which I think could be the issue. It says --flash_size "4MB"

But at the end of the file we have:

```
FLASH_INFO=16MB (128Mb)
```

which I think comes from FLASH_DEF=16M

plerup, 2025-02-09:
Which board is this?

everslick, 2025-02-09:
A ESP32-WROVER-E-N16R8 module.

plerup, 2025-02-09:
OK, see it now, fix coming

plerup, 2025-02-09:
Fixed in latest commit

everslick, 2025-02-11:
Tested and confirmed working! Thank you very much, for this lightning fast fix!

closed 2025-02-11 by everslick