plerup/makeEspArduino #208: Extern C Error

diabl0w, 2025-04-06:
when compiling a project that uses FastLED - I get dozens of errors involving extern C. I already opened an issue with FastLED (https://github.com/FastLED/FastLED/issues/1814), but they told me its an issue with the build environment. Do you know how this can be fixed:
```
/home/user/src/MCUs/ESP/toolchains/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/reent.h:9:1: note: 'extern "C"' linkage started here
    9 | extern "C" {
      | ^~~~~~~~~~
In file included from /home/user/src/MCUs/ESP/temp/makeEspArduino/libs/FastLED/src/fx/time.h:6,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/time.h:414,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/tools/sdk/lwip2/include/lwipopts.h:3560,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/tools/sdk/lwip2/include/lwip/init.h:40,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/cores/esp8266/wl_definitions.h:72,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:28,
                 from /home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:25:
/home/user/src/MCUs/ESP/temp/makeEspArduino/libs/FastLED/src/fl/callback.h:12:1: error: template with C linkage
   12 | template<typename ...Args>
      | ^~~~~~~~
In file included from /home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:25:
/home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:27:1: note: 'extern "C"' linkage started here
   27 | extern "C" {
      | ^~~~~~~~~~
In file included from /home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ArduinoOTA/ArduinoOTA.cpp:22:
/home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/include/UdpContext.h: In member function 'void UdpContext::seek(size_t)':
/home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/include/UdpContext.h:224:9: error: 'assert' was not declared in this scope
  224 |         assert(isValidOffset(pos));
      |         ^~~~~~
/home/user/src/MCUs/ESP/toolchains/esp8266/libraries/ESP8266WiFi/src/include/UdpContext.h:33:1: note: 'assert' is defined in header '<cassert>'; did you forget to '#include <cassert>'?
   32 | #include <AddrList.h>
  +++ |+#include <cassert>
   33 | #include <PolledTimeout.h>
```

diabl0w, 2025-04-08:
This occurs with an empty project that only has `#include <FastLED.h>` so it should be easy to reproduce

diabl0w, 2025-04-13:
chatgpt helping me figure out the issue, i think i guided it to start giving me relevant information:
```
Based on the output, here’s the key observation:

Observation:
Your global include directories contain your user libraries — notably, the FastLED library (and all its subdirectories) is included. For example, you have this line in your output:

/home/user/Arduino/libraries/FastLED/src

This means that when core system files (such as those in the ESP8266 core) are compiled, they’re also seeing your FastLED headers. In the normal Arduino build process, only the sketch files get to see these extra libraries, but your current Makefile is applying them globally.

Why is that a problem?
Some ESP8266 core files (like cont_util.cpp) are written in C/C++ with a strict expectation on the header context. When the FastLED headers (which contain C++ templates) are pulled in by core files—because they’re in a global include path—it causes errors like:

error: template with C linkage

This happens because FastLED is not meant to be part of the core's compilation unit
```

diabl0w, 2025-04-13:
After months of dealing with this, i FINALLY figured out a workaround. @plerup do you think you can turn this simple workaround into a more generic fix:

```
--- a/makeEspArduino.mk
+++ b/makeEspArduino.mk
@@ -211,6 +211,11 @@ ifneq ($(MAKECMDGOALS),clean)
 -include $(SRC_LIST)
 endif
 
+# Example for a specific source file that needs FastLED:
+ifneq (,$(findstring FastLED,$(USER_SRC)))
+    USER_INC_DIRS += /home/user/Arduino/libraries/FastLED/src
+endif
+
 ifeq ($(suffix $(SKETCH)),.ino)
   # Use sketch copy with correct C++ extension
   SKETCH_CPP = $(BUILD_DIR)/$(notdir $(SKETCH)).cpp
diff --git a/tools/find_src.pl b/tools/find_src.pl
index 8d8e554..227fcea 100644
--- a/tools/find_src.pl
+++ b/tools/find_src.pl
@@ -155,8 +155,10 @@ foreach (keys %src_files) {
 print "USER_INC_DIRS = ";
 # Keep order
 foreach (@search_dirs) {
-  print "$_ " if $inc_dirs{$_};
+  #print "$_ " if $inc_dirs{$_};
+  print "$_ " if $inc_dirs{$_} && !($_ =~ /FastLED/);  # Add condition to avoid global inclusion of FastLED
 }
 print "\n";
 print "USER_SRC = ", join(" ", sort(keys %src_files)), "\n";
```

plerup, 2025-04-20:
Well, as far as I can see the problem is that this library has its own overrides of some system include files.
Have you tried using the EXCLUDE_DIRS variable?

diabl0w, 2025-04-20:
> Well, as far as I can see the problem is that this library has its own overrides of some system include files. Have you tried using the EXCLUDE_DIRS variable?

I can't really EXCLUDE_DIRS, because then I wouldn't be able to use the library right?

plerup, 2025-04-20:
You could try just adding the conflicting sub directories to EXCLUDE_DIRS (note reg exp).
As far as I can see they are referenced with sub paths in the #include statements

diabl0w, 2025-04-20:
> You could try just adding the conflicting sub directories to EXCLUDE_DIRS (note reg exp). As far as I can see they are referenced with sub paths in the #include statements

this works to get past the compiling step (`EXCLUDE_DIRS="FastLED/src/.+`), but then fails at the linking step:
```
xtensa-lx106-elf/bin/ld: /home/user/src/MCUs/ESP/esp8266_ws2812/build/user_obj.ar(main.cpp.o): in function `_ZN19ClocklessControllerILi3ELi20ELi50ELi30EL6EOrder66ELi0ELb0ELi85EE4initEv':
/home/user/src/MCUs/ESP/libraries/FastLED/src/platforms/esp/8266/clockless_esp8266.h:37: undefined reference to `_ZN14CLEDController20clearLedDataInternalEi'
```

plerup, 2025-04-20:
Try EXCLUDE_DIRS="FastLED/src/fx"

diabl0w, 2025-04-20:
> Try EXCLUDE_DIRS="FastLED/src/fx"

if i do that, then i still get extern C errors from the "fl" subdirectory (my original post only showed 1 error but there were dozens more that i excluded for brevity)

so then, if i also exclude fl subdirectory, I will get linkage errors

so the "fl" subdirectory causes one issue if excluded (linkage error) and another issue if not excluded (extern C)

plerup, 2025-04-20:
Can you send me a simple example which gets this problem?

diabl0w, 2025-04-20:
> Can you send me a simple example which gets this problem?

clone: 
https://github.com/FastLED/FastLED.git

 create main.ino:
```
#include <FastLED.h>
void loop () {}
```

compile:
for esp8266 using makeEspArduino

plerup, 2025-04-21:
Fixed in latest commit. New variable EXCLUDE_INC which can be used for excluding include directories. FastLED added to this by default as it is a known problem.

diabl0w, 2025-04-25:
> Fixed in latest commit. New variable EXCLUDE_INC which can be used for excluding include directories. FastLED added to this by default as it is a known problem.

Im so sorry, we had a sudden turnover of employees at work and got really busy... this works fantastically! thank you!!!!
quick question, what would the syntax be for this? especially if i want to add something hardcoded into the find_src.pl

i am getting this:
```
/home/user/src/MCUs/ESP/libraries/pubsubclient/tests/src/lib/Arduino.h:21: warning: "PROGMEM" redefined
   21 | #define PROGMEM
```
and so i thought i could do:
```
$exclude_inc_match .= ($exclude_inc_match ? "|" : "") . '/FastLED/src/|/pubsubclient/tests/'; # Known problem
```
but that did not work, still get same error

diabl0w, 2025-04-25:
nevermind, i think that situation is where i would use the regular EXCLUDE... thanks again! closing